ankitdhall/lidar_camera_calibration #2: cross_sensor_calibration/launch does not exist...

zachkendrick, 2017-06-23:
Hi,

Really cool ROS package! I like the approach using AR tags. I built the package but I am having issues running it. The directory cross_sensor_calibration/launch does not exist. I was wondering if you could add this or clarify where the new launch file is. Please note that I am fairly new to ROS so I am not an expert in launching ROS nodes.

Thanks!

zachkendrick, 2017-06-23:
I think in the tutorial you meant to put `$ roslaunch lidar_camera_calibration find_transform.launch` as the command to run the ros node.

ankitdhall, 2017-06-24:
thank you @zachkendrick
There were some typos in the README.md file, which have been edited.
Yes, the command to run the node is `roslaunch lidar_camera_calibration find_transform.launch`

closed 2017-06-24 by ankitdhall