ankitdhall/lidar_camera_calibration #5: lidar input node

zachkendrick, 2017-06-26:
Do you happen to have two nodes/code that publish the lidar and camera data to the topics:

camera_frame_topic: /frontNear/left/image_raw
velodyne_topic: /velodyne_points

Thanks for all the help and the quick responses!

ankitdhall, 2017-06-27:
Hi @zachkendrick 
These topics are published by the respective drivers for the LiDAR and the camera. So when you run the device drivers (usually in the form of a ros node) after connecting the LiDAR and camera to your computer, the nodes will publish these topics (and a host of other topics as well).

zachkendrick, 2017-06-27:
Hi @ankitdhall 
I figured that was the case so I got a rosbag with data of the type sensor_msgs::PointCloud2 and topic /velodyne_points. However, when I launch find_transform.launch and I play the rosbag, find_transform hangs at `[ INFO] [1498576976.830800350]: Reading CameraInfo from configuration file`. I know that the rosbag is publishing to the topic /velodyne_points because I can do a `$ rostopic echo /velodyne_points` and the data gets printed out to the terminal. Is there any reason you can think of as to why find_transform is hanging? Thank you so much for all the help!

ankitdhall, 2017-06-28:
@zachkendrick yes, you need both the image and pointcloud messages to be published together. And yes, you need to have the topics synchronized together (have the same timestamp).

zachkendrick, 2017-06-29:
I have both /velodyne_points and /image_raw publishing data. However, the program hangs when the memo8 opencv window pops up. It won't detect the aruco tags that I have printed. I tried using your image that you have in the readme and it detected the aruco tags but it still hung. Do you have any idea what might be going on? It seems that the callback function isn't being called because the print statement "velodyne scan received at ___" isn't printing out. When I look at the output of the /velodyne_points topic it shows data though... any advice would be much appreciated.

ankitdhall, 2017-06-29:
It could probably be because the topic names are different from those in the readme. Another reason could be that the timestamps do not match. Also, you mentioned using the image from the readme, you must have used some other pointcloud possibly leading to a timestamp problem.

closed 2017-07-04 by ankitdhall

KleinYuan, 2017-11-09:
@zachkendrick 
Hey man, I met the same issue as you guys.
Mono8 windows show up and then hangs.
It looks like you made it work eventually and what's the issue then?

KleinYuan, 2017-11-09:
@zachkendrick @ankitdhall 
I met the same issue that it hangs at
```
init done
opengl support available
```
with Mono8 poping up.

Any hints on how to fix that?
Camera is pulished to `/camera/image_raw` and I changed the remap accordingly in the launch file.

Is it an OpenCV or QT issue?

jediofgever, 2018-01-24:
Please , can someone tell me how to synchronize two topics. I can’t figure this out. When I run the command in usage section, it stucks.

karnikram, 2018-01-24:
@jediofgever Can you describe your setup? How are you getting the messages?

jediofgever, 2018-01-25:
Hi @karnikram,  I installed the package successfully. I have published data from LIDAR and ZED camera successfully. I did not do the physical setup, I just wanted to see if I am able to start the process with  'roslaunch lidar_camera_calibration find_transform.launch'. I changed the topics name in 'lidar_camera_calibration.yaml' file to correct ones. But still it will just hang saying.  '[ INFO] [1498576976.830800350]: Reading CameraInfo from configuration file'.

I found @zachkendrick  suggestion,  syncrhonize timestamps might be solution of this issue but I dont know what part of code should be modified to do this.

Also, do you think not setting up marker board properly can cause this problem ? 
Thank you

karnikram, 2018-01-25:
If you're playing the messages from a bag file, the marker messages and lidar messages will have different timestamps.
Do `rosparam set use_sim_time true` before playing your bag file. And add `--clock` to your rosbag play command.

jediofgever, 2018-01-25:
Thank you @karnikram . I dont use rosbag file.  I directly take data from Velodyne-16 LIDAR in real time. I will update this comment after trying your suggestion.

jediofgever, 2018-01-26:
@karnikram I tried the command `rosparam set use_sim_time true` still the same. It stucks just as before

karnikram, 2018-01-26:
I don't understand your issue exactly.

Are you getting both the velodyne messages and the marker_6dof messages?
You need to run aruco_mapping to get the latter messages published. Are you
doing that?

jediofgever, 2018-01-26:
here is what I get from `rostopic list` :
`/aruco_markers
/aruco_poses
/diagnostics
/rosout
/rosout_agg
/scan
/tf
/tf_static
/velodyne_nodelet_manager/bond
/velodyne_nodelet_manager_cloud/parameter_descriptions
/velodyne_nodelet_manager_cloud/parameter_updates
/velodyne_nodelet_manager_driver/parameter_descriptions
/velodyne_nodelet_manager_driver/parameter_updates
/velodyne_nodelet_manager_laserscan/parameter_descriptions
/velodyne_nodelet_manager_laserscan/parameter_updates
/velodyne_packets
/velodyne_points
/zed/depth/camera_info
/zed/depth/depth_registered
/zed/depth/depth_registered/compressed
/zed/depth/depth_registered/compressed/parameter_descriptions
/zed/depth/depth_registered/compressed/parameter_updates
/zed/depth/depth_registered/compressedDepth
/zed/depth/depth_registered/compressedDepth/parameter_descriptions
/zed/depth/depth_registered/compressedDepth/parameter_updates
/zed/depth/depth_registered/theora
/zed/depth/depth_registered/theora/parameter_descriptions
/zed/depth/depth_registered/theora/parameter_updates
/zed/joint_states
/zed/left/camera_info
/zed/left/camera_info_raw
/zed/left/image_raw_color
/zed/left/image_raw_color/compressed
/zed/left/image_raw_color/compressed/parameter_descriptions
/zed/left/image_raw_color/compressed/parameter_updates
/zed/left/image_raw_color/compressedDepth
/zed/left/image_raw_color/compressedDepth/parameter_descriptions
/zed/left/image_raw_color/compressedDepth/parameter_updates
/zed/left/image_raw_color/theora
/zed/left/image_raw_color/theora/parameter_descriptions
/zed/left/image_raw_color/theora/parameter_updates
/zed/left/image_rect_color
/zed/left/image_rect_color/compressed
/zed/left/image_rect_color/compressed/parameter_descriptions
/zed/left/image_rect_color/compressed/parameter_updates
/zed/left/image_rect_color/compressedDepth
/zed/left/image_rect_color/compressedDepth/parameter_descriptions
/zed/left/image_rect_color/compressedDepth/parameter_updates
/zed/left/image_rect_color/theora
/zed/left/image_rect_color/theora/parameter_descriptions
/zed/left/image_rect_color/theora/parameter_updates
/zed/odom
/zed/point_cloud/cloud_registered
/zed/rgb/camera_info
/zed/rgb/camera_info_raw
/zed/rgb/image_raw_color
/zed/rgb/image_raw_color/compressed
/zed/rgb/image_raw_color/compressed/parameter_descriptions
/zed/rgb/image_raw_color/compressed/parameter_updates
/zed/rgb/image_raw_color/compressedDepth
/zed/rgb/image_raw_color/compressedDepth/parameter_descriptions
/zed/rgb/image_raw_color/compressedDepth/parameter_updates
/zed/rgb/image_raw_color/theora
/zed/rgb/image_raw_color/theora/parameter_descriptions
/zed/rgb/image_raw_color/theora/parameter_updates
/zed/rgb/image_rect_color
/zed/rgb/image_rect_color/compressed
/zed/rgb/image_rect_color/compressed/parameter_descriptions
/zed/rgb/image_rect_color/compressed/parameter_updates
/zed/rgb/image_rect_color/compressedDepth
/zed/rgb/image_rect_color/compressedDepth/parameter_descriptions
/zed/rgb/image_rect_color/compressedDepth/parameter_updates
/zed/rgb/image_rect_color/theora
/zed/rgb/image_rect_color/theora/parameter_descriptions
/zed/rgb/image_rect_color/theora/parameter_updates
/zed/right/camera_info
/zed/right/camera_info_raw
/zed/right/image_raw_color
/zed/right/image_raw_color/compressed
/zed/right/image_raw_color/compressed/parameter_descriptions
/zed/right/image_raw_color/compressed/parameter_updates
/zed/right/image_raw_color/compressedDepth
/zed/right/image_raw_color/compressedDepth/parameter_descriptions
/zed/right/image_raw_color/compressedDepth/parameter_updates
/zed/right/image_raw_color/theora
/zed/right/image_raw_color/theora/parameter_descriptions
/zed/right/image_raw_color/theora/parameter_updates
/zed/right/image_rect_color
/zed/right/image_rect_color/compressed
/zed/right/image_rect_color/compressed/parameter_descriptions
/zed/right/image_rect_color/compressed/parameter_updates
/zed/right/image_rect_color/compressedDepth
/zed/right/image_rect_color/compressedDepth/parameter_descriptions
/zed/right/image_rect_color/compressedDepth/parameter_updates
/zed/right/image_rect_color/theora
/zed/right/image_rect_color/theora/parameter_descriptions
/zed/right/image_rect_color/theora/parameter_updates
/zed/zed_wrapper_node/parameter_descriptions
/zed/zed_wrapper_node/parameter_updates`
Is there any missing topics ?

karnikram, 2018-01-26:
/lidar_camera_calibration_rt messages are missing.
The aruco_mapping package will publish those when it detects the markers.

RTFM.

jediofgever, 2018-01-26:
whoa I got rtfm now.

here is my setup, 
[image]
and this is from LIDAR 
[image]
And this is what ZED camera see
[image]
Do you see a not proper set up ?

dkhanna511, 2020-11-05:
So I am doing the exact same thing as you/ I have not made the setup and I want to know if things work. But when I run aruco_mapping.launch file, it gets stuck at  " .830800350]: Reading CameraInfo from configuration file". With all the discussion followed, can I conclude that this is only because I haven't made the setup and the camera and lidar are just not able to detect the markers?