ankitdhall/lidar_camera_calibration #6: lidar_camera_calibration_rt topic has no publisher

zachkendrick, 2017-06-29:
I think I have found the reason why it is hanging but I don't know how to fix it. The callback function in find_velodyne_points.cpp subscribes to the topic "lidar_camera_calibration_rt" but this topic has no publisher when I do `$ rostopic info /lidar_camera_calibration_rt`. I am assuming the topic `/aruco_poses` should publish to it but that topic is currently publishing a completely different type (aruco_mapping/ArucoMarker). Any thoughts?

zachkendrick, 2017-06-29:
I found the correct aruco_mapping files that contain the correct publisher here:

http://mirror-eu.docs.ros.org/latest-lts/api/lidar_camera_calibration/html/dir_03b35d6ddc8b0dcc071773c41473f2e5.html

I think the link to the aruco_mapping in the README is incorrect... let me know if I am right or wrong.

ankitdhall, 2017-06-30:
@zachkendrick the dependencies are available in the `lidar_camera_calibration/dependencies` folder. The readme mentions that `aruco_mapping` has been modified slightly and publishes the `lidar_camera_calibration_rt` topic. You will find more information in the `Setup` section in the readme.

closed 2017-06-30 by ankitdhall

villanuevab, 2018-06-14:
I checked the velodyne points timestamps and camera timestamps, but the approximate time synchronizer's callback is never called. I am assuming this is related to this issue. When I run rostopic list, I don't see any `lidar_camera_calibration_rt` topic.

chenwinki, 2019-04-26:
@villanuevab   Hi, How did you deal with "ApproximateTimeSync callback can't be call" this problem? I met the same problem. Hope for reply.