ankitdhall/lidar_camera_calibration #7: Coordinate system of rotation and translation?

zachkendrick, 2017-07-01:
Can you give me coordinate system of the translation/lidar? Thanks for all the help by the way! I have gotten the package to run to completion. Unfortunately, I don't think the results of the translation are accurate. Any suggestions?

karnikram, 2017-07-01:
Hey, 
I've been using the package too and I've been getting quite accurate results. Double check the intrinsic calibration of your camera and your marker measurements in the marker_coordinates.txt file.

If you're using the velodyne driver, the orientation follows the standard ROS body frame convention ie the X axis of the device would point forward, Y towards the left and Z upwards.

ankitdhall, 2017-07-04:
@zachkendrick 
I hope the comment by @karnikram helps.
The rotation and translation is such that it transforms all the points in the LiDAR frame to the camera frame; so the camera is the reference/origin.

closed 2017-07-04 by ankitdhall