ankitdhall/lidar_camera_calibration #8: Calibration Issues...

zachkendrick, 2017-07-05:
Hi,

I am really struggling to get good results. Any help would be much appreciated. @ankitdhall and @karnikram here are some typical results from the calibration:

```
Distance between the corners:
0.525242
0.501325
0.519984
0.509987
Num points is:8
Number of points: 8
  2.35791  -0.184031  -0.138072
-0.185836   0.504375 -0.0946709
-0.135297 -0.0896248  0.0302841
Rotation matrix:
  0.999998  0.00105429 -0.00192951
-0.00107425    0.999946   -0.010374
0.00191847    0.010376    0.999944
Rotation in Euler angles:
179.952
-179.903
-179.419
Translation:
0.00544582
 0.0184382
-0.00412293
RMSE: 0.00636279
Rigid-body transformation:
  0.999998  0.00105429 -0.00192951  0.00544582
-0.00107425    0.999946   -0.010374   0.0184382
0.00191847    0.010376    0.999944 -0.00412293
         0           0           0           1
--------------------------------------------------------------------
After 81 iterations
--------------------------------------------------------------------
Average translation is:
 0.0271066
 0.0151644
-0.00466677
Average rotation is:
  0.999998  0.00105429 -0.00192951
-0.00107425    0.999946   -0.010374
0.00191847    0.010376    0.999944
Average transformation is:
  0.999998  0.00105429 -0.00192951   0.0271066
-0.00107425    0.999946   -0.010374   0.0151644
0.00191847    0.010376    0.999944 -0.00466677
         0           0           0           1
Average RMSE is: 0.0141243
RMSE on average transformation is: 0.0228186
```
The camera and lidar are separated by about 40cm horizontally but the translation values are at least an order of magnitude smaller than that. I double checked the intrinsic calibration values and the other config files. Here they are:

**Config_file.txt**

```
640 480
-4.5 4.5
-4.0 4.0
-4.0 4.0
0.005
2
0
635.519348 0.0        306.008153 0.0
0.0        634.956848 249.757176 0.0
0.0        0.0        1.0        0.0
100
```

**marker_coordinates.txt**

```
2
50.0
50.0
3.8
0.5
20.6
50
51.6
3.5
0.3
20.6
```

**find_transform.launch**

```
<?xml version="1.0"?>
<launch>
 <!-- <param name="/use_sim_time" value="true"/> -->
 

 <!-- ArUco mapping -->
<node pkg="aruco_mapping" type="aruco_mapping" name="aruco_mapping" output="screen">
   <!--<remap from="/image_raw" to="/frontNear/left/image_raw"/>-->

   <param name="calibration_file" type="string" value="$(find aruco_mapping)/data/liveCam.ini" />
   <param name="num_of_markers" type="int" value="2" />
   <param name="marker_size" type="double" value="0.206"/>
   <param name="space_type" type="string" value="plane" />
   <param name="roi_allowed" type="bool" value="false" />

 </node>

 <rosparam command="load" file="$(find lidar_camera_calibration)/conf/lidar_camera_calibration.yaml" />
 <node pkg="lidar_camera_calibration" type="find_transform" name="find_transform" output="screen">
 </node>

 <node pkg="lidar_input" type="lidar_input_node" name="lidar_input_node" output="screen">
 </node>

 <node pkg="camera_input" type="camera_input_node" name="camera_input_node" output="screen">
 </node>
</launch>
```

Thank you for all your help so far. I really appreciate it. If you have any insight into why this might not be working, please please please let me know as soon as possible.

Thanks!

zachkendrick, 2017-07-07:
Small Bug: On line 119 of Corners.cpp the path is missing a `/` the path should be `pkg_loc + "/conf/points.txt"`. This is what was causing my bug. It seems to work now! Thanks for all the help!

closed 2017-07-09 by ankitdhall

sssoubl, 2018-10-30:
@zachkendrick:Did you get good results! I meet the same problem with you。