ankitdhall/lidar_camera_calibration #9: Points.txt number of points

zachkendrick, 2017-07-06:
The README says that the file points.txt should contain a number at the begining of the file that corresponds to `num_of_sensors*(num_of_markers*points_per_board)`. I looked at the code and it seems that this number should just be `(num_of_markers*points_per_board)` . For the example targets that are in the README the number of points would then be 8 rather than the suggested 16. Is this correct?

zachkendrick, 2017-07-07:
Small Bug: On line 119 of Corners.cpp the path is missing a `/` the path should be `pkg_loc + "/conf/points.txt"`. This is what was causing my bug. It seems to work now! Thanks for all the help!

ankitdhall, 2017-07-08:
Thanks for pointing out the typo. :+1:

closed 2017-07-08 by ankitdhall