ankitdhall/lidar_camera_calibration #12: calibration process stuck

wzhouuu, 2017-08-15:
I followed the instruction and running the calibration toolbox in ubuntu14.04. But the process is stuck at 'Reading CameraInfo from configuration file' and not continue. I'm not sure it's my settings problem or my laptop is not powerful enough?
Btw, the velodyne and camera nodes are running in another machine. I access them via ROS Master. Thanks!
[image]

wzhouuu, 2017-08-15:
I figured out the problem was didn't uncomment the sections in launch file. Now I'm having another problem where the toolbox seems can't subscribe to the velodyne. Again, the velodyne node is running on another machine. rostopic list shows both camera topic and velodyne topic
[image]

ankitdhall, 2017-08-15:
Assuming the networking part is working correctly. There are a couple of things you can look into.
[1] Make sure the timestamps for each topic are synchronized
[2] `lidar_camera_calibration.yaml` should have the correct names of the topics being published by the velodyne and camera, so that the node can subscribe to them and begin the calibration process.

Also, you can have a look at issue #5 for [1].

closed 2017-08-21 by wzhouuu

jediofgever, 2018-01-23:
Hi, I hit the same error. I tried to do step 1 but unfourtunately I couldnt find a solution. Can you explain me, how to do synchronization between this two topics ?
Thank you.

dkhanna511, 2020-11-05:
Hi,
I am having the same issue of getting stuck at "Reading CameraInfo from configuration file". Can you tell me which part needs to be uncommented in the launch file? That would be of great help. Thanks.