ankitdhall/lidar_camera_calibration #13: pool per

leica8244, 2017-08-15:
the data in poins.txt is as following:
8
-0.56811 -0.473328 1.79659
-0.213204 -0.134026 1.7816
-0.543555 0.218787 1.78348
-0.915342 -0.132713 1.8172
0.145415 -0.450564 1.79215
0.56126 -0.191263 1.78309
0.289001 0.229009 1.78225
-0.113636 -0.0443698 1.79478
-0.485469 -0.327932 1.47909
-0.144692 0.0051091 1.53701
-0.471128 0.350685 1.47055
-0.811904 0.0176443 1.41263
0.124684 -0.314695 1.50046
0.524897 -0.0503131 1.51868
0.260503 0.350273 1.51351
-0.139709 0.0858913 1.49529
it seems that the points' coordinates of the camera are much different from those of the lidar.
at the same time, the result of my calibratioin seems not so correct. because my camera is just next to the lidar.
(
Average transformation is: 
  0.996523       0.00304858     -0.0832566    0.176845
-0.00350322    0.99998           -0.0053152    0.124334
 0.0832387       0.00558839     0.996514      -0.28119
         0               0                      0                    1
)
can you help my figure out what may be wrong ?and how to make it more acurate?

ankitdhall, 2017-08-15:
There are a few things you can check.
[1] Check the values you have entered in the `config_file.txt`
[2] `marker_coordinates.txt` has the correct measurements of the board on which the ArUco marker is stuck.
[3] marker size in `find_transform.launch` is correctly entered
You could probably re-read the README to make sure that you haven't missed anything.

leica8244, 2017-08-17:
thank you for that, it is the launch file which i forget to modify. 
it's now correctly calibrated.

leica8244, 2017-08-17:
Now we have gotten a good R&t matrix, and we try to fuse two point cloud.
Even when we don't move the camera at all ,the final-fused point cloud seems not coincide so good.
We put the lidar and camera together,and the t is 0.03,0.05,0.07, this seems good enough.
And the final fused point cloud look like this .
How can i get as a good result as yous?
Thank you in advance!
[image]

ankitdhall, 2017-08-17:
One of the things you could look at is ensuring that the intrinsic calibration of the cameras is good. Better the intrinsic calibration will lead to more precise fusion of the point clouds. Also, make sure that the cameras remain in the same place, even the slightest movements can cause discrepancy in the fused point cloud. A good idea is to fix them on a bar or some rigid object, so they do not change their relative positions.

closed 2017-08-22 by ankitdhall

wenjunj, 2019-05-31:
@leica8244 你好，很高兴看到你是上海交通大学的，我也是上海交大的，想咨询你一点问题，我标定出来的结果RMSE很小，但是我实际配准激光和图片时效果不理想，请问大概是什么原因的，是否方便添加微信18818207380，非常感谢，希望有机会向你请教！
Num points is:8
Number of points: 8
   0.559911   -0.074347 -0.00786292
 -0.0554011    0.200673  -0.0119373
 -0.0318813  0.00789918 0.000815446
Rotation matrix: 
  0.999208  -0.021123  0.0337249
 0.0239192   0.996112 -0.0847847
-0.0318029  0.0855243   0.995828
Rotation in Euler angles: 
1.37139
1.82261
4.90902
Translation: 
 -0.185703
 0.0968357
-0.0996245
RMSE: 0.0198393
Rigid-body transformation: 
  0.999208  -0.021123  0.0337249  -0.185703
 0.0239192   0.996112 -0.0847847  0.0968357
-0.0318029  0.0855243   0.995828 -0.0996245
         0          0          0          1
--------------------------------------------------------------------
After 100 iterations
--------------------------------------------------------------------
Average translation is:
 -0.176358
 0.0460298
-0.0925272
Average rotation is:
  0.999232 -0.0183505  0.0346194
 0.0205445   0.997731  -0.064123
-0.0333642   0.064785   0.997341
Average transformation is: 
  0.999232 -0.0183505  0.0346194  -0.176358
 0.0205445   0.997731  -0.064123  0.0460298
-0.0333642   0.064785   0.997341 -0.0925272
         0          0          0          1
Final rotation is:
 0.0354151  -0.999218  0.0175548
-0.0641067 -0.0198011  -0.997747
  0.997314  0.0342099 -0.0647578
Final ypr is:
 2.07551
 -1.6441
-0.48601
Average RMSE is: 0.0216582
RMSE on average transformation is: 0.0366648

zhangyuran-gg, 2024-11-19:
> ```
>  @leica8244 你好，很高兴看到你是上海交通大学的，我也是上海交大的，想咨询你一点问题，我标定出来的结果RMSE很小，但是我实际配准激光和图片时效果不理想，请问大概是什么原因的，是否方便添加微信18818207380，非常感谢，希望有机会向你请教！
> ```
> 
> ## 点数为：8 点
> 数： 8
> 0.559911 -0.074347 -0.00786292
> -0.0554011 0.200673 -0.0119373
> -0.0318813 0.00789918 0.000815446
> 旋转矩阵：
> 0.999208 -0.021123 0.0337249
> 0.0239192 0.996112 -0.0847847
> -0.0318029 0.0855243 0.995828
> 欧拉角旋转：
> 1.37139
> 1.82261
> 4.90902
> 平移：
> -0.185703
> 0.0968357
> -0.0996245
> RMSE： 0.0198393
> 刚体变换：
> 0.999208 -0.021123 0.0337249 -0.185703
> 0.0239192 0.996112 -0.0847847 0.0968357
> -0.0318029 0.0855243 0.995828 -0.0996245
> 0 0 0 1
> ## 迭代 100 次后
> 平均翻译为： -0.176358 0.0. 0460298 -0.0925272 平均旋转为： 0.999232 -0.0183505 0.0346194 0.0205445 0.997731 -0.064123 -0.0333642 0.064785 0.997341 平均变换为： 0.999232 -0.0183505 0.0346194 -0.176358 0.0205445 0.997731 -0.064123 0.0460298 -0.0333642 0.064785 0.997341 -0.0925272 0 0 0 1 最终旋转为： 0.0354151 -0.999218 0.0175548 -0.0641067 -0.0198011 -0.997747 0.997314 0.0342099 -0.0647578 最终 ypr 为：2.07551 -1.6441 -0.48601 平均 RMSE 为：0.0216582 平均变换为 RMSE 为：0.0366648

你好，现在这个问题有解决了吗