ankitdhall/lidar_camera_calibration #14: how to understand calibration result

wzhouuu, 2017-08-21:
Hi, I tried twice for this calibration toolbox. I got two different answers for the same camera-velodyne setup. Our velodyne is located ~5cm above the camera, and the camera is pointing 15 degree downwards. Here is the result we got:
​Second test:
After 100 iterations
--------------------------------------------------------------------
Average translation is:
-0.220729
-0.273677
4.82629
Average rotation is:
 0.905049  0.419485 0.0701358
  0.40611 -0.901351   0.15047
 0.126337 -0.107699 -0.986124
Average transformation is:
 0.905049  0.419485 0.0701358 -0.220729
  0.40611 -0.901351   0.15047 -0.273677
 0.126337 -0.107699 -0.986124   4.82629
        0         0         0         1
Average RMSE is: 0.325415
RMSE on average transformation is: 0.331159

First test:
After 100 iterations
--------------------------------------------------------------------
Average translation is:
0.422593
0.831942
 3.90745
Average rotation is:
 0.900951   0.36973 -0.227127
  0.22701 -0.847701 -0.479447
-0.369801  0.380398  -0.84767
Average transformation is: 
 0.900951   0.36973 -0.227127  0.422593
  0.22701 -0.847701 -0.479447  0.831942
-0.369801  0.380398  -0.84767   3.90745
        0         0         0         1
Average RMSE is: 0.346004
RMSE on average transformation is: 0.353328

Is there any documentation on how to read this result? I visualized rotation matrix with ROS tf, but neither of them seems correct. Do you have any suggestions what I should do to get good results? Will this toolbox work for sensor setup like ours? Btw, our camera is 100degree FoV and Velodyne is 16 beams. Really appreciate your help!

chinitaberrio, 2017-08-22:
I don't know, sorry

ankitdhall, 2017-08-22:
The results can be read as estimates that have been computed over multiple iterations. An average of the translation and rotation are computed over the N iterations. The translation is a 3x1 vector with x, y, z components and the rotation values are in the form of a 3x3 matrix.  

The average transformation is just the stacked matrix `[R t; 0 0 0 1]` (4x4 matrix).
The RMSE is the root mean squared error between the 3D points viewed from the camera and laser scanner after applying the transformation.  

`lidar_camera_calibration` was specifically made for a setup that had (multiple) cameras (both stereo and monocular) and a laser scanner with 16 scan-lines. You can have a look at the fusion results of point clouds (available in the README) from stereo cameras after calibrating using this method.  

There are a couple of things you should have a look at to make sure your setup is correct (these are also mentioned in the README):
[1] Check the values you have entered in the config_file.txt
[2] marker_coordinates.txt has the correct measurements of the board on which the ArUco marker is stuck.
[3] marker size in find_transform.launch is correctly entered
[4] ensure that the intrinsic calibration of the camera is of high quality; better calibration will give more accurate estimates  
You could probably re-read the README to make sure that you haven't missed anything. Also you can have a look at #13   
For more technical details you can refer to [this](http://arxiv.org/pdf/1705.09785).

closed 2017-08-30 by ankitdhall