ankitdhall/lidar_camera_calibration #16: Few questions for setup

robo7man, 2017-08-30:
Setup: Velodyne and Point Grey Cameras in Stereo setup.

1. Is there an ideal distance between camera-lidar and the markers. Right now my markers are detected only when close( 3ft-4ft away). 

2. There seems to be different values of projection matrix in conf/config_file.txt 
```
611.651245 0.0        642.388357 0.0
0.0        688.443726 365.971718 0.0
0.0        0.0        1.0        0.0
```

and the corresponding data/zed_left_uurmi.ini in aruco_mapping. 
```
projection
6.8230856098675258e+02 0 6.2940856933593750e+02 0 
0 6.8230856098675258e+02 3.8939595031738281e+02 0 
0 0 1 0
```
Is it intentional or is there something more to this?

Thanks!

wzhouuu, 2017-08-30:
I think you need to adjust line 2 to line4 in config_file.txt. They are x-,x+; y-,y+ and z-, and z+. Those numbers limit the cubic space that will be detected by the sensor. Try increase them and see the results?

ankitdhall, 2017-08-31:
[1] During the experiments, the markers were being detected up to a distance of 2m. That being said, maybe with larger marker sizes, the detection range increases; you could check that in the ArUco marker documentation.  

[2] The difference in the values is just a coincidence, the files/values must have been altered while testing with multiple (different) cameras.

closed 2017-09-07 by ankitdhall

villanuevab, 2018-07-25:
For [2], if in config_file.txt we set the `use_camera_info_topic?` to `0` (use intrinsic parameters from config_file.txt), are the parameters in the .ini file used anywhere?

karnikram, 2018-07-26:
The .ini file is used only by the aruco_mapping package.