ankitdhall/lidar_camera_calibration #20: Calibration is wrong

emilaz, 2017-10-25:
Hello,
the results of the provided calibration tool seem to be very off. I tried running it a couple of times but the end result is often as can be seen below:

[image]

To reproduce this result, I am uploading an appropriate bag file and all the needed config files:
[test.bag.zip](https://github.com/ankitdhall/lidar_camera_calibration/files/1415520/test.bag.zip)
[conf.zip](https://github.com/ankitdhall/lidar_camera_calibration/files/1415531/conf.zip)
[find_transform.launch.zip](https://github.com/ankitdhall/lidar_camera_calibration/files/1415541/find_transform.launch.zip)
[zed_left_uurmi.ini.zip](https://github.com/ankitdhall/lidar_camera_calibration/files/1415539/zed_left_uurmi.ini.zip)

and the corresponding results I got:
[results3.txt](https://github.com/ankitdhall/lidar_camera_calibration/files/1415521/results3.txt)

I am thankful for any suggestion on how to improve the result.

karnikram, 2017-10-25:
Hey, 

I haven't looked into your data completely, but it seems like your markers are positioned incorrectly.

The smaller valued marker should always be on the left!

emilaz, 2017-10-26:
Hey, 
Marker #904 is on the left, #905 on the right.

karnikram, 2017-10-26:
My bad. 905 looked like 903 on my system.

Anyway, I ran your data and I seem to be getting the right result.

Transformation , T :

T =  [0.996554 -0.00467748  -0.0828156   0.0894906
  0.0109373    0.997101   0.0752965    0.156528
  0.0822233  -0.0759428    0.993716    0.388588
          0           0           0           1];

[image]

emilaz, 2017-11-01:
So yours is the average transformation after 100 iterations? I don't understand what the final rotation is telling me, since its values always seem to differ greatly from the average rotation.
Also, would you be able to provide the visualization tool you used for the above graphic?

Greets,
Emil

karnikram, 2017-11-01:
Yes, I used the average transformation.

The script can be found here - #19

karnikram, 2017-11-09:
Sorry, missed your other question.
final_rot = average_rot * initial_rot

initial_rot is usually (pi/2,-pi/2,0) depending on your configuration

closed 2017-11-20 by ankitdhall

SinghiDivyanshu, 2020-01-24:
hi @karnikram  @emilaz ,
i,m stuck after getting the rigid body transformation matrix, now don't know how to visualize the output (point clound and image in same frame). here i'm using the monocular camera.