ankitdhall/lidar_camera_calibration #21: Met Poor Result..

won4113, 2017-10-31:
I run with correct intrinsic camera parameter, marker configuration file, fixed launch files, but it shows so poor result(RMSE : 0.37~~~)

[image]

It seems some error with our running method with this package, but I can't find any error...

I would really appreciate it if anyone could help with our calibration.
Below files are our configuration files, launch file and rosbag file of our experiment.

[config_file.zip](https://github.com/ankitdhall/lidar_camera_calibration/files/1429648/config_file.zip)
[lidar_camera_calibration.zip](https://github.com/ankitdhall/lidar_camera_calibration/files/1429649/lidar_camera_calibration.zip)
[marker_coordinates.zip](https://github.com/ankitdhall/lidar_camera_calibration/files/1429650/marker_coordinates.zip)
[find_transform.zip](https://github.com/ankitdhall/lidar_camera_calibration/files/1429652/find_transform.zip)
[zed_left_uurmi.zip](https://github.com/ankitdhall/lidar_camera_calibration/files/1429653/zed_left_uurmi.zip)
https://drive.google.com/open?id=0B_ZpibQh0KW6NjYwOTJKUEp1aUU
(rosbag file)

karnikram, 2017-11-01:
Hi,

Hold the markers as shown in the README. Each corner must be distinctly visible in the point cloud.

It becomes very hard to find the corner locations in the lidar's point cloud when the markers are held horizontally like you have.

won4113, 2017-11-02:
Hi. Thanks for the answer.
I understand that corner points do not look good on point cloud because of horizontally located markers.
But I think some problem occurs with my method, not on our markers.
I have found that no matter how I choose the point (even if I choose the wrong point) the error is about 0.37....

sangych, 2018-06-11:
Hi，I used the config files and rosbag you provided. but the filtered pointcloud was not shown ,what's wrong? thank you for help
I just run as this:
rosbag play rosbag_filename
roslaunch lidar_to_camera_calibration find_transform.launch

Chrislzy1993, 2018-10-17:
@sangych , 我觉得应该是topic同步的问题， /velodyne_points时间戳是录制bag包时的时间戳， 但是/lidar_camera_calibration_rt是运行/aruco_mapping节点发出的时间戳， 两者根本对不上， 所以都没进入callback 函数

xinwf, 2018-10-25:
@Chrislzy1993 也就是说，如果激光雷达的时间戳和aruco_mapping的时间戳对不上不会调用callback函数？激光雷达自身没有时间戳怎么办啊

uglyxjsxl, 2018-11-05:
> @sangych , 我觉得应该是topic同步的问题， /velodyne_points时间戳是录制bag包时的时间戳， 但是/lidar_camera_calibration_rt是运行/aruco_mapping节点发出的时间戳， 两者根本对不上， 所以都没进入callback 函数

请教一下,我用雷神的C16和一个普通单目摄像头,每次都会出现"Failed to find match for field 'ring'. "的警告,每次都报错,exit code -11,所有的话题消息都是对,请问可能是什么原因?
[image]
偶尔会弹出检测到的窗口,但是迭代到第二次就报同样的错误..

Chrislzy1993, 2018-11-05:
> @Chrislzy1993 也就是说，如果激光雷达的时间戳和aruco_mapping的时间戳对不上不会调用callback函数？激光雷达自身没有时间戳怎么办啊

你可以看看代码， 里面用到了ros的message_filter中的Time Synchronizer， 这个就是基于timestamp同步的， 如果你使用的激光雷达没有时间戳的话，那么你需要修改程序，改成你自己的同步方式，但是比较好奇的是你用的是哪种激光雷达，为什么会没有时间戳，还是你暂时没发现时间戳

Chrislzy1993, 2018-11-05:
> 请教一下,我用雷神的C16和一个普通单目摄像头,每次都会出现"Failed to find match for field 'ring'. "的警告,每次都报错,exit code -11,所有的话题消息都是对,请问可能是什么原因?
> [image]
> 偶尔会弹出检测到的窗口,但是迭代到第二次就报同样的错误..

如果你的雷达用的是veleody的话，那么程序基本不需要改动，如果不是的话，需要改动一些地方，主要是在提取marker纸板边缘上点的那个程序， 在仔细看看源程序，你会发现的，good luck

Geo-fortune, 2018-11-08:
如果不用rosbag包，偶尔能进入“GetConer.cpp”的代码中，能显示“cloud”和“polygen”的窗口，但是鼠标无法在上面进行选择。
在使用rosbag包的时候，我也遇到了exit code -11的错误，但是这个错误是在aruco _mapping代码中 “// Calculate TF between two markers”的时候“aruco_mapping.cpp”代码的“474”行出现的。

xinwf, 2018-11-15:
> > @Chrislzy1993 也就是说，如果激光雷达的时间戳和aruco_mapping的时间戳对不上不会调用callback函数？激光雷达自身没有时间戳怎么办啊
> 
> 你可以看看代码， 里面用到了ros的message_filter中的Time Synchronizer， 这个就是基于timestamp同步的， 如果你使用的激光雷达没有时间戳的话，那么你需要修改程序，改成你自己的同步方式，但是比较好奇的是你用的是哪种激光雷达，为什么会没有时间戳，还是你暂时没发现时间戳

校准程序里面的时间戳用的都是ROS系统时间的时间戳，没有问题，今天有些进展，其实是作者的文档写的不够详细，是非常不详细，它给的视频我看的时候没有声音不知道为什么，关于opencv的waitkey函数我一开始就不知道什么意思，所以 @Geo-fortune 才说，鼠标无法选择，这地方不是用鼠标选，鼠标单击一次后，要按一下键盘，按的键与waitkey中的值对应。

xinwf, 2018-11-15:
> 请教一下,我用雷神的C16和一个普通单目摄像头,每次都会出现"Failed to find match for field 'ring'. "的警告,每次都报错,exit code -11,所有的话题消息都是对,请问可能是什么原因?
> [image]
> 偶尔会弹出检测到的窗口,但是迭代到第二次就报同样的错误..

许瑨？

uglyxjsxl, 2018-11-16:
> 许瑨？

巧啊，哈哈

yosungho, 2019-10-09:
I got the poor result, either. (RMSE 0.25) Anyone succeed to get a good result?