ankitdhall/lidar_camera_calibration #25: Fatal error: Assertion failed

blutjens, 2017-11-28:
Hello, 
the calibration process has died after 74 iterations. The displayed error is the following.
Are you familiar with the error and could you please explain what this error means and how it is caused?
`[FATAL] [1511892092.578293398]: ASSERTION FAILED`
`	file = .../src/lidar_camera_calibration/include/lidar_camera_calibration/Find_RT.h`
`	line = 277`
`	cond = marker_info.size()/7 == num_of_marker_in_config`
Thanks !

blutjens, 2017-12-11:
I assume the error is produced during finding the marker corners from the lidar data (step 4). The program flow on each iteration how I've understood it is:
1. Retrieve a new frame from the lidar
2. Filter it, based on the big static bounding box from the config file
3. Assign the 3D points on the edges, based on the by-mouse-click-input bounding boxes
4. Fit lines onto the marked edges to find the marker's corners.
5. Find and update avg transformation from lidar to camera frame.

The assertion failed error, may be caused by the program not being able to find matching corners for the marked points. So, the solution would be to assure, that the by-mouse-click-input bounding boxes only take in values from the actual marker. Am I right?

karnikram, 2017-12-16:
Hello,
Your understanding of the program's flow is right.

I think the error occured becuse the program couldn't find the required number of corners in that particular iteration. The points could have lied outside your bounding boxes in that iteration.

closed 2018-03-11 by ankitdhall

marijanp, 2019-12-18:
I had the same assertion exception and it took me a while to find the problem. What I didn't notice was, that the marker I've used was to small and the aruco mapping node didn't detect the marker. Therefore the marker_info was empty. As a solution one has to use a bigger marker, such that the aruco mapping node is able to detect it.