ankitdhall/lidar_camera_calibration #28: Kabsch or ICP

blutjens, 2017-12-11:
In your paper, you mention the results for ICP (iterative approximation) and Kabsch (closed-form solution) algorithm. 

1. Why does ICP outperform Kabsch in the RMSE? Is the error in Kabsch caused by averaging the rotation matrices?

2. Does the code use ICP or Kabsch algorithm? Is there a way of switching in between both?

Thanks!

karnikram, 2017-12-16:
A1) I'm unsure about it myself. Maybe @ankitdhall could help.
A2) This program uses Kabsch. The ICP implementation hasn't been included in this repository. But I thnk PCL's implementation of ICP was used!

ankitdhall, 2017-12-17:
Kabsch is used because point correspondences are known. A reason for error could be error in measurements of point coordinates. The current code does not use ICP.

closed 2018-01-10 by ankitdhall

lkqnaruto, 2022-11-20:
> Kabsch is used because point correspondences are known. A reason for error could be error in measurements of point coordinates. The current code does not use ICP.

Can you explain what does "point correspondences" mean? Thank you