ankitdhall/lidar_camera_calibration #29: When launch 'find_transform.launch', only refresh Txyz and Rxyz of Markers constantly.

PanZhichen, 2017-12-21:
When I configure everything according to the REDME written in your GitHub , and roslaunch the 'find_transform.launch', then I got a strange result as I pasted below:

`[molmc-chy:~/velodyne_ws] 28s $ roslaunch lidar_camera_calibration find_transform.launch 
... logging to /home/molmc-chy/.ros/log/0ed1a0b4-e597-11e7-abf6-0021ccce43d0/roslaunch-molmcchy-QiTianM4550-D601-979.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:38896/

SUMMARY
========

PARAMETERS
 * /aruco_mapping/calibration_file: /home/molmc-chy/v...
 * /aruco_mapping/marker_size: 0.18
 * /aruco_mapping/num_of_markers: 2
 * /aruco_mapping/roi_allowed: False
 * /aruco_mapping/roi_height: 480
 * /aruco_mapping/roi_width: 752
 * /aruco_mapping/roi_x: 0
 * /aruco_mapping/roi_y: 0
 * /aruco_mapping/space_type: plane
 * /lidar_camera_calibration/camera_frame_topic: /camera/image_raw
 * /lidar_camera_calibration/camera_info_topic: /camera/camera_info
 * /lidar_camera_calibration/velodyne_topic: /velodyne_points
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /
    aruco_mapping (aruco_mapping/aruco_mapping)
    find_transform (lidar_camera_calibration/find_transform)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[aruco_mapping-1]: started with pid [1003]
process[find_transform-2]: started with pid [1004]
[ INFO] [1513782265.402906924]: Calibration file path: /home/molmc-chy/velodyne_ws/src/aruco_mapping/data/ptgrey.ini
[ INFO] [1513782265.402995617]: Number of markers: 2
[ INFO] [1513782265.403044328]: Marker Size: 0.18
[ INFO] [1513782265.403073615]: Type of space: plane
[ INFO] [1513782265.403111941]: ROI allowed: 0
[ INFO] [1513782265.403139203]: ROI x-coor: 0
[ INFO] [1513782265.403166121]: ROI y-coor: 0
[ INFO] [1513782265.403194306]: ROI width: 32654
[ INFO] [1513782265.403220837]: ROI height: 4272766
[ INFO] [1513782265.405128762]: Calibration data loaded successfully
Size of the frame: 752 x 480
Limits: -1.5 to 1.5
Limits: -1.5 to 1.5
Limits: -1.5 to 1.5
Number of markers: 2
Intensity threshold (between 0.0 and 1.0): 1
useCameraInfo: 0
Projection matrix: 
[722.22699, 0, 396.83157, 0;
  0, 724.25177, 245.65433, 0;
  0, 0, 1, 0]
MAX_ITERS: 100
[ INFO] [1513782265.438624388]: Reading CameraInfo from configuration file
66=(168.113,201.139) (117.111,260.515) (55.3148,210.179) (107.151,150.758) Txyz=-0.658777 -0.0930726 1.67338 Rxyz=-0.537329 1.75076 -1.79857 
77=(527.799,188.93) (474.491,244.359) (419.016,189.193) (473.023,134.821) Txyz=0.177901 -0.130918 1.68745 Rxyz=-0.672243 1.69652 -1.80348 
[ INFO] [1513782265.783152542]: First marker with ID: 66 detected
[ WARN] [1513782265.786862766]: TF to MSG: Quaternion Not Properly Normalized
66=(168.051,201.149) (117.164,260.534) (55.2879,210.181) (107.144,150.758) Txyz=-0.658857 -0.0930523 1.67352 Rxyz=-0.536725 1.75233 -1.7963 
77=(527.753,188.969) (474.498,244.32) (418.889,189.18) (473.076,134.887) Txyz=0.177839 -0.130921 1.68762 Rxyz=-0.67746 1.69446 -1.80724 
[ INFO] [1513782265.891261512]: Velodyne scan received at 1.51378e+09
[ INFO] [1513782265.891307601]: marker_6dof received at 1.51378e+09

Initial Rot 0.000796274    -0.999999 -0.000796274
           0  0.000796274           -1
           1  0.000796274  6.34053e-07
66 -0.658777 -0.0930726 1.67338 -0.537329 1.75076 -1.79857 77 0.177901 -0.130918 1.68745 -0.672243 1.69652 -1.80348 
[ INFO] [1513782265.893154385]: iteration number: 0

---------Moving on to next marker--------
66=(168.059,201.162) (117.11,260.535) (55.2527,210.139) (107.145,150.785) Txyz=-0.65902 -0.0930747 1.67384 Rxyz=-0.535501 1.75028 -1.797 
77=(527.734,189.02) (474.462,244.329) (418.946,189.123) (473.167,134.796) Txyz=0.177871 -0.130906 1.68714 Rxyz=-0.669608 1.69471 -1.80134 
66=(168.105,201.131) (117.075,260.514) (55.3207,210.173) (107.163,150.705) Txyz=-0.658476 -0.0930691 1.67254 Rxyz=-0.540196 1.75179 -1.79922 
77=(527.844,188.929) (474.486,244.333) (418.937,189.165) (473.07,134.82) Txyz=0.177844 -0.130933 1.68707 Rxyz=-0.678422 1.69711 -1.80545 
66=(168.087,201.175) (117.226,260.374) (55.2333,210.285) (107.143,150.793) Txyz=-0.659902 -0.0932084 1.67636 Rxyz=-0.526221 1.7461 -1.79708 
77=(527.898,188.899) (474.463,244.331) (418.946,189.123) (473.142,134.821) Txyz=0.177917 -0.130992 1.68731 Rxyz=-0.682443 1.69949 -1.80451 
66=(168.079,201.155) (117.131,260.515) (55.3069,210.189) (107.136,150.735) Txyz=-0.658763 -0.0930562 1.67328 Rxyz=-0.537594 1.75257 -1.7965 
77=(527.899,188.926) (474.478,244.345) (419.016,189.193) (473.091,134.752) Txyz=0.177934 -0.130927 1.68709 Rxyz=-0.675419 1.70117 -1.80028 
66=(168.117,201.146) (117.073,260.525) (55.3364,210.155) (107.169,150.737) Txyz=-0.658583 -0.0930623 1.67287 Rxyz=-0.539795 1.75157 -1.79877 
77=(527.862,188.957) (474.49,244.357) (419.105,189.281) (473.207,134.571) Txyz=0.178031 -0.130836 1.68657 Rxyz=-0.658211 1.70799 -1.78504 
66=(168.076,201.203) (117.145,260.511) (55.2496,210.223) (107.148,150.794) Txyz=-0.659179 -0.0930375 1.67435 Rxyz=-0.533089 1.74945 -1.79672 
77=(527.941,188.852) (474.52,244.387) (419.052,189.229) (473.249,134.617) Txyz=0.17802 -0.130877 1.6862 Rxyz=-0.666335 1.70853 -1.78896 
66=(168.119,201.179) (117.1,260.536) (55.2809,210.218) (107.152,150.757) Txyz=-0.658676 -0.0929982 1.67308 Rxyz=-0.536713 1.75028 -1.79851 
77=(527.83,188.935) (474.566,244.433) (419.041,189.217) (473.33,134.501) Txyz=0.17786 -0.13068 1.68401 Rxyz=-0.6496 1.70332 -1.78422 
`

PanZhichen, 2017-12-21:
the 'aruco_mapping' can run normally, but when launch 'find_transform.launch', there are no points in "point" window.

sangych, 2018-06-13:
@PanZhichen I was confused about this issue too, how did you make it up? Thanks a lot

PanZhichen, 2018-06-18:
@sangych Each time you pick up a point by left-click, pressing any key on your keyboard will help.

ykk136, 2020-12-22:
Did you use usb cameras, such as PointGrey USB3? Or cameras using Ethernet to transport data? I found that my USB3 camera is not synchonized with Velodyne lidar, so I can't get what I want..