ankitdhall/lidar_camera_calibration #30: After Calibration, I got 'Average rotation' and 'Final rotation'. Which one should be used?

PanZhichen, 2017-12-26:
[image]

Thanks for your help!

karnikram, 2017-12-26:
Hi,
If you check the code you'll find,
`Eigen::Matrix3d final_rotation = rotation_avg * lidarToCamera;`
lidarToCamera is the initial configurable rotation between the lidar and the camera.
So if you need to fuse raw data from both the sensors, use final_rotation.

PanZhichen, 2017-12-27:
@karnikram Thanks for your reply!
Yes, I have found this line of code. 
So whether that's means the '**Final rotation**' should be used when I want to transform original points received by lidar to camera frame? In this case, I think the value hung under '**Final ypr is**' in the screenshot above should be close to the initial configurable ratation value set in '**config_file.txt**'. 
In my customized situation I set the lidar and camera both pointing forward, so I think the 'ypr' value(2.94.. -1.50.. -1.40..) should be close to the value (1.57 -1.57 0) that I have set before hand. And the first two values are appropriate since the third one vary widely, so any thing to do to solve this problem? Or my understanding is wrong? Please give me some advise, thank you very much!

karnikram, 2017-12-27:
Use the final rotation value if you want to transform the raw lidar coordinates to the camera's frame.
The final rotation and initial rotation are in fact quite close. You can have different sets of euler angles correspond to the same rotation matrix. 
You can convert both sets of angles to their corresponding rotation matrices and see for yourself. The rotation specified in the config file is (r,p,y) while the final rotation is (y,p,r), so take care of that.

closed 2018-03-11 by ankitdhall

villanuevab, 2018-03-29:
Is y,p,r yaw, pitch, roll? I get very different values when calculating these directly from the final rotation matrix (using these formulas for y, p, r: http://planning.cs.uiuc.edu/node103.html). Also, is there a final translation value calculated? Or only average...?

karnikram, 2018-03-29:
Hi,
Yes they are the yaw pitch roll angles. It is possible to get different values since euler angle representations are not unique.

In this case, the final and average translation values are the same. This is because we first apply the rotation to match the orientation of the frames, and then the translation. So the final (or total) transformation would be [R_final | t_avg]