ankitdhall/lidar_camera_calibration #32: marker_6dof.h directory not found

shivam-chandhok, 2018-01-01:
[ 21%] Linking CXX shared library /home/shivam/catkin_ws/devel/lib/libaruco.so
[ 21%] Built target aruco
/home/shivam/catkin_ws/src/lidar_camera_calibration/src/find_velodyne_points.cpp:34:50: fatal error: lidar_camera_calibration/marker_6dof.h: No such file or directory
compilation terminated.
lidar_camera_calibration/CMakeFiles/find_transform.dir/build.make:62: recipe for target 'lidar_camera_calibration/CMakeFiles/find_transform.dir/src/find_velodyne_points.o' failed
make[2]: *** [lidar_camera_calibration/CMakeFiles/find_transform.dir/src/find_velodyne_points.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:2999: recipe for target 'lidar_camera_calibration/CMakeFiles/find_transform.dir/all' failed
make[1]: *** [lidar_camera_calibration/CMakeFiles/find_transform.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

shivam-chandhok, 2018-01-03:
why am I getting this error

karnikram, 2018-01-13:
Please read the README properly.

aruco_ros and aruco_mapping need to be installed before you can install this package. You can find these dependencies inside the dependencies folder.

closed 2018-03-11 by ankitdhall

villanuevab, 2018-03-26:
Can you please write more descriptive installation instructions regarding `aruco_mapping` and `aruco_ros`? I cloned `aruco_ros` via to `catkin_ws/src` and copied https://github.com/ankitdhall/lidar_camera_calibration/tree/master/dependencies/aruco_mapping to `catkin_ws/src` as well. I am getting a similar error: 

```
CMakeFiles/Makefile2:27094: recipe for target 'lidar_camera_calibration/CMakeFiles/find_transform.dir/all' failed
make[1]: *** [lidar_camera_calibration/CMakeFiles/find_transform.dir/all] Error 2
```

How do we install the necessary dependencies?

karnikram, 2018-03-27:
Hi,

Yes the aruco packages need to be installed before the main package can be installed.

And thanks for the suggestion, we'll update the README about this.