ankitdhall/lidar_camera_calibration #34: How could I do 2D-3D correspondence calibration?

xiaoshuliu, 2018-01-08:
Hi, I have read your paper, and I wonder could I just do the 2D-3D correspondence calibration without ArUco markers? Is the code for 2D-3D in this repo? Thanks.

karnikram, 2018-01-13:
The code has not been included in this repo.

And I don't think the 2D-3D approach worked as well as this one.

OmarBarakat1995, 2018-01-19:
@karnikram I've read the paper but couldn't get how 3D features are obtained in case of monocular . Is it because the boards are planar, thus all edge points are having same depth?

karnikram, 2018-01-19:
Aruco gives us the 3d pose of the marker relative to the camera directly. And the coordinates of the board's corner points relative to the aruco marker are known since the board's dimensions are known. So a simple matrix multiplication can be done to obtain the coordinates of the corner points in the camera's frame.

OmarBarakat1995, 2018-01-19:
@karnikram  Thank you .. But I am considered with the depth (3rd dimension) in case of monocular camera not stereo. let me thank you again and if I am misunderstanding something please refer me to a read .

xiaoshuliu, 2018-01-19:
@karnikram How does ArUco give us the 3d pose? Is it the same principle as the chess board, but just use a different library?

karnikram, 2018-01-19:
@OmarBarakat1995  Stereo images are not being used for pose estimation here, only monocular. Aruco markers or any fiducial markers can be used for monocular pose estimation. Refer to [this](http://www.uco.es/investiga/grupos/ava/sites/default/files/GarridoJurado2014.pdf) for an explanation of the arcuo library.

OmarBarakat1995, 2018-02-05:
@karnikram we have configured the files to calibrate our monocamera with vlp lidar... On launching the find_transform launch file we get this error. 
[image]
please note that we used the camera calibrator which provided us with .yaml camera parameters file and amother .text file , we couldn't find an explicit .ini file so we renamed the text file tochange its extension and provided its path to the launch file parameter default value.

closed 2018-03-11 by ankitdhall