ankitdhall/lidar_camera_calibration #35: find_transform.launch doesn't respond after detecting first marker (if the second is in the scene)

OmarBarakat1995, 2018-02-06:
[image]

blutjens, 2018-02-13:
@OmarBarakat1995 Check that your calibration path in find_transforms.launch points to your camera's intrinsic parameter file (e.g. 'zed_left_uurmi.ini'). You can receive the values for your intrinsic camera by calling $ rostopic echo /your_camera/camera_info.

villanuevab, 2018-05-14:
I am getting the same issue (except no warning for incorrect calibration). I encountered the issue when trying to use a non-velodyne lidar.

closed 2018-06-29 by ankitdhall

villanuevab, 2018-07-25:
I am encountering the same issue. Is the correct calibration file needed to run? I recognize it's needed for a good result, but I didn't think it would interrupt the calibration process.