ankitdhall/lidar_camera_calibration #38: Links to video are broken

va9abund, 2018-03-07:
Hello.

Please update links to Youtube you provided.

Best regards, Alex.

ankitdhall, 2018-03-08:
@va9abund the video links seem to be fine. Could you try them again or let me know which one seems broken to you?

va9abund, 2018-03-08:
The first one in usage section (https://youtu.be/SiPGPwNKE-Q), in "Fusion using..."(https://youtu.be/AbjRDtHLdz0) and "Calibrating cameras ... " (https://youtu.be/Om1SFPAZ5Lc).

ankitdhall, 2018-03-08:
They seem to work fine for me. :disappointed:

va9abund, 2018-03-08:
Could you write me the name of the youtube chanel and close the issue? Thanks :)

ankitdhall, 2018-03-09:
Here is the link to the playlist [link](https://www.youtube.com/playlist?list=PLoeQ9AU3xgsrsj9iS2sMMUnHZTLuCejeA).

closed 2018-03-10 by ankitdhall