ankitdhall/lidar_camera_calibration #40: How to use average translation and YPR angles with ROS from LIDAR and Camera data fusion

jediofgever, 2018-03-30:
Hi, 
I run the package, and got results as shown below, 
[image]

I would like to visualize Lidar point clods in Zed camera frame. So I do a static coordinate transformation with this command

`rosrun tf static_transform_publisher 0 -0.09122 -0.04545 -0.0432 1.48563 -1.5026 0.13215  zed_left_camera velodyne 100`

However result look non relevant as shown below

[image]
What I am doing wrong ? Is there another way to use transformation matrix(4X4) in ROS
below are my files and rosbag
[zed_left_uurmi.ini](https://github.com/jediofgever/HackerRank_Solutions_Python/blob/master/zed_left_uurmi.ini)
[config_file.txt](https://github.com/ankitdhall/lidar_camera_calibration/files/1863551/config_file.txt)

[marker_coordinates.txt](https://github.com/ankitdhall/lidar_camera_calibration/files/1863553/marker_coordinates.txt)
Rosbag : https://drive.google.com/open?id=1HKqOUeIacSARwwUcCV6CU83fjTWr3OwI

karnikram, 2018-03-30:
Hi,

What is the white colored plane in your visualization? That doesn't seem like the point cloud from your zed.

You could alternatively run `stereo_image_proc` for obtaining the zed's point cloud and then fuse them together in rviz like you've done.

 Another way of visualizing the results is described in #19 and #20

jediofgever, 2018-03-30:
There is the section in your code , `pointcloud_fusion.` But it is designed for fusing 2 camera point clouds. However I just want to fuse Lıdar and zed camera point clouds.  Can you tell me if I still can use that code for this ?

karnikram, 2018-03-30:
If you used the ZED SDK to obtain that depth cloud then that might be the problem since they use a different axes convention.

You can modify `pointcloud_fusion` for your case by directly giving the final transform, provided you have the respective pcd files.

jediofgever, 2018-03-30:
Right,  I use zed sdk. Last question, I am not sure if the problem is the way I fuse data or the calibration parameters are not correct ?,
Do the calibration parameters my program found seems reasonable ?

karnikram, 2018-03-31:
Your calibration parameters do seem correct to me.

The way you're verifying them in rviz is also correct, but like I said zed uses a different axes convention and not the standard convention. According to their convention x points forward, y towards the left, and z upward, which is the same as the velodyne's. The two initial rotations (about x and y) are not needed in that case.

But this package uses the standard ROS convention for the camera axes, which needs the initial rotations, and hence there's a mismatch in your output. I suggest you use stereo_image_proc and visualize it again.

jediofgever, 2018-03-31:
I tried to use `stereo_image_proc`, however I stuck on getting topics not synchronized error. Can you share your launch file that you use. such as `roslaunch stereo_image_proc your.launch` Thank you

karnikram, 2018-04-02:
I guess you're missing `camera_info` messages.
I do not have a launch file for `stereo_image_proc`, I just run it from the command line with the `__ns` argument.

jediofgever, 2018-04-02:
Thank you for all the help. ! Got it working 
[image]

closed 2018-04-02 by jediofgever

karnikram, 2018-04-03:
Can you post your solution here for reference?

jediofgever, 2018-04-03:
Here is the command I pass to get over of it ;
`rosrun stereo_image_proc stereo_image_proc __ns:=zed _approximate_sync:=True`