ankitdhall/lidar_camera_calibration #41: Cannot detect aruco markers

LiShuaixin, 2018-03-30:
Hi, thanks for your sharing. I'm trying to calibrate the velodyne lidar with a Kinect2 camera . But I cannot detect the aruco markers, which makes the program stuck. I tried to run the aruco_mapping node only, and it also dosen't work.

Why the marker cannot be detected? I don't know what's wrong. Please help!!!!!!!!!!!!!!!!!! T T!
[image]

jediofgever, 2018-03-30:
Make sure you remap your camera topic to /image_raw, also the size of aruco marker need to be in meters

LiShuaixin, 2018-03-31:
@jediofgeve thanks for your reply. I checked my launch file and topics, but didn't find the problem. Here is my launch file and rqt_graph of nodes and topics. Could you find what's wrong?  
[image]
[image]

jediofgever, 2018-03-31:
Your launch file looks fine, and topics also seem ok. I also could not figure out why it is not detecting. If you are not running bag file I would suggest you to change this line `  <param name="/use_sim_time" value="true"/> ` to `  `. If that does not solve it then Only thing I can imagine is your camera parameters at `kinect2_righ.ini`

LiShuaixin, 2018-03-31:
Here is my .ini file in which my camera intrinsic parameters have been included. All things look fine, but the markers still could not be detected.

`# Prosilica camera intrinsics

[image]

width
1920

height
1080

[prosilica]

camera matrix
1078.7590613742789 0.00000 949.45707686180253 
0.00000 1081.0857157086082 529.69624694028789 
0.00000 0.00000 1.00000

distortion
0.091093900790104432 -0.14051805018202065 -0.0013933224835513888 -0.0033859633806729498 0.064918028653487753

rectification
1.00000 0.00000 0.00000
0.00000 1.00000 0.00000
0.00000 0.00000 1.00000 

projection
1078.7590613742789 0.00000 949.45707686180253 0.00000
0.00000 1081.0857157086082 529.69624694028789 0.00000
0.00000 0.00000 1.00000 0.00000

`

jediofgever, 2018-03-31:
I could not reproduce your issue with both ZED camera and XBOX Kinect. I used your ini file and it successfully detected markers,
[image]

 maybe  rebuilding aruco_mapping package can handle

LiShuaixin, 2018-03-31:
could you share your launch file with me?

LiShuaixin, 2018-03-31:
what's the topic do you remap?

 ```
 <!-- ArUco mapping -->
  <node pkg="aruco_mapping" type="aruco_mapping" name="aruco_mapping" output="screen">
    <remap from="/image_raw" to="/kinect2/hd/image_color_rect"/>

    <param name="calibration_file" type="string" value="$(find aruco_mapping)/data/kinect2_right.ini" /> 
    <param name="num_of_markers" type="int" value="2" />
    <param name="marker_size" type="double" value="0.50"/>
    <param name="space_type" type="string" value="plane" />
    <param name="roi_allowed" type="bool" value="false" />
    <!--
    <param name="roi_x" type="int" value="0" /> 
    <param name="roi_y" type="int" value="0" />
    <param name="roi_width" type="int" value="640" /> 
    <param name="roi_height" type="int" value="480" /> 
  -->

  </node>  

```

jediofgever, 2018-03-31:
can you do a` rostopic list` while your camera node is running, what else topics do you have ?

The topic I remap is  `/camera/rgb/image_color`

LiShuaixin, 2018-03-31:
Here is all topics:

/kinect2/bond
/kinect2/hd/camera_info
/kinect2/hd/image_color
/kinect2/hd/image_color/compressed
/kinect2/hd/image_color_rect
/kinect2/hd/image_color_rect/compressed
/kinect2/hd/image_depth_rect
/kinect2/hd/image_depth_rect/compressed
/kinect2/hd/image_mono
/kinect2/hd/image_mono/compressed
/kinect2/hd/image_mono_rect
/kinect2/hd/image_mono_rect/compressed
/kinect2/hd/points
/kinect2/qhd/camera_info
/kinect2/qhd/image_color
/kinect2/qhd/image_color/compressed
/kinect2/qhd/image_color_rect
/kinect2/qhd/image_color_rect/compressed
/kinect2/qhd/image_depth_rect
/kinect2/qhd/image_depth_rect/compressed
/kinect2/qhd/image_mono
/kinect2/qhd/image_mono/compressed
/kinect2/qhd/image_mono_rect
/kinect2/qhd/image_mono_rect/compressed
/kinect2/qhd/points
/kinect2/sd/camera_info
/kinect2/sd/image_color_rect
/kinect2/sd/image_color_rect/compressed
/kinect2/sd/image_depth
/kinect2/sd/image_depth/compressed
/kinect2/sd/image_depth_rect
/kinect2/sd/image_depth_rect/compressed
/kinect2/sd/image_ir
/kinect2/sd/image_ir/compressed
/kinect2/sd/image_ir_rect
/kinect2/sd/image_ir_rect/compressed
/kinect2/sd/points
/rosout
/rosout_agg

jediofgever, 2018-03-31:
change `/kinect2/hd/image_color_rect` to `/kinect2/hd/image_color` and let us know if that fixes

LiShuaixin, 2018-03-31:
I have tried all these topics: 
```
/kinect2/hd/image_color
/kinect2/hd/image_color/compressed
/kinect2/hd/image_color_rect
/kinect2/hd/image_color_rect/compressed
```

But no one could success.

LiShuaixin, 2018-03-31:
And here is the problem: 
```
X server found. dri2 connection failed! 
DRM_IOCTL_I915_GEM_APERTURE failed: invalid parameters
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [22]
param: 4, val: 0
```

jediofgever, 2018-03-31:
This looks like an issue with your cuda on your NVIDIA. I suggest you to reinstall latest CUDA 9.1 as well as latest Nvidia driver (should be 390).   you can find how to install CUDA on your machine  [Here](https://askubuntu.com/questions/799184/how-can-i-install-cuda-on-ubuntu-16-04)

LiShuaixin, 2018-04-01:
@jediofgever  Thank you for your help. I'll have a try! My nvida driver is the version of 385. Maybe it's too old. But last time when I try to update the graphic driver, my OS broke down. This time I 'll be more careful and when I done that I'll let you know. Thanks again for your help and guidance

LiShuaixin, 2018-04-16:
I've installed CUDA under your instruction, but the marker still cannot be detected. I debug the aruco_mapping node, and find that the problem occured after 220 line in aruco_mapping.cpp, which is "Detector.detect(input_image,temp_markers,aruco_calib_params_,marker_size_);" The detect function is the one defined in aruco::MarkerDetector class. What can be the problem occured in this function? If both of the environment for marker detection and the marker itself have some specific requirements?

dbruggner, 2018-04-18:
I have the same problem with latest master. I cannot detect any Aruco markers (tried with different marker sizes).

LiShuaixin, 2018-04-19:
@dbruggner Have you tried using aruco detection function in Opencv3? I can detect markers using detectMarkers() function in opencv3, while cannot detect any markers with different marker sizes either.

ZhengPan2014, 2018-06-19:
@LiShuaixin Have you solved this problem, I got the same problem that aruco_mapping can't detect marker?

ZhengPan2014, 2018-06-19:
@LiShuaixin Have you calibrated lidar+camera successfully with this package, I got the problem that no marker detected too?

tkkhuu, 2018-10-11:
@LiShuaixin have you been able to detect the markers? I trying the package and also unable to detect anything

ericzzj1989, 2019-10-17:
> @dbruggner Have you tried using aruco detection function in Opencv3? I can detect markers using detectMarkers() function in opencv3, while cannot detect any markers with different marker sizes either.

Have you been able to detect the markers? I trying the package and also unable to detect anything

ericzzj1989, 2019-10-17:
Hi, @LiShuaixin . Have you been able to detect the markers? I trying the package and also unable to detect anything

adugin, 2019-11-03:
You MUST have significant white border around each marker. The right one doesn't have any. The left one probably has insufficient border (the bottom side).