ankitdhall/lidar_camera_calibration #42: Does it work with other lidar types?

villanuevab, 2018-04-03:
I tried using another lidar device and updated the velodyne topic in `lidar_camera_calibration` https://github.com/ankitdhall/lidar_camera_calibration/blob/master/conf/lidar_camera_calibration.yaml#L4 . I am getting the following error: 
```
[ INFO] [1522777695.370229735]: First marker with ID: 10 detected
[ WARN] [1522777695.382916640]: TF to MSG: Quaternion Not Properly Normalized
[ INFO] [1522777695.432211706]: Velodyne scan received at 1.52278e+09
[ INFO] [1522777695.432235629]: marker_6dof received at 1.52278e+09
Failed to find match for field 'ring'.

Initial Rot 0.000796274    -0.999999 -0.000796274
           0  0.000796274           -1
           1  0.000796274  6.34053e-07
10=(1107.69,1012.23) (1024.24,1102.09) (935.85,1021.21) (1020.12,931.427) Txyz=0.111969 0.849601 2.85558 Rxyz=-0.620609 1.73495 -1.76494 
20=(1498.92,1023.35) (1407.57,1114.76) (1310.73,1023.5) (1400.71,932.817) Txyz=0.816052 0.817916 2.70597 Rxyz=-0.78969 1.8706 -1.74944 
10=(1107.74,1012.18) (1024.28,1102.05) (935.82,1021.17) (1020.11,931.481) Txyz=0.111959 0.849404 2.85501 Rxyz=-0.620971 1.73789 -1.76304 
20=(1498.92,1023.39) (1407.6,1114.73) (1310.82,1023.57) (1400.71,932.755) Txyz=0.816282 0.818081 2.70652 Rxyz=-0.78935 1.86932 -1.7504 
[find_transform-2] process has died [pid 18752, exit code -11, cmd /.../devel/lib/lidar_camera_calibration/find_transform __name:=find_transform __log:=/cyngn_log/9a3596ec-3763-11e8-bf5e-1c1b0d9bb6f8/find_transform-2.log].
log file: /log/9a3596ec-3763-11e8-bf5e-1c1b0d9bb6f8/find_transform-2*.log
``` 

I am able to calibrate when using a Velodyne lidar. The only difference that I made to cause this error is use a different lidar and different topic name. Both use PointCloud2.msg.

villanuevab, 2018-04-03:
The lidar driver I have does not publish ring information. I think this is the issue.

closed 2018-04-03 by villanuevab

Chrislzy1993, 2018-10-12:
hi, i have the same issue with you , i comment some code . and  the package can work , but in cloud window , it is wrong .  it seems like the package didn't detect the edge in cloudpoints

oaix, 2018-10-31:
I use the lidar of robosense. Only neet to modify some codes.

pedghz, 2019-02-11:
Hi everyone! did you manage to use other LiDARS like robosense with this package?
@oaix could you please how did you manage to use robosense LiDAR? I am getting an error saying
> Failed to find match for field 'ring'.
> terminate called after throwing an instance of 'std::bad_alloc'
> what(): std::bad_alloc

and after that the process dies and generates this output:
> [find_transform-2] process has died [pid 31697, exit code -6, cmd /home/pedram/catkin-ws/devel/lib/lidar_camera_calibration/find_transform __name:=find_transform > __log:=/home/pedram/.ros/log/75cc0b6e-2d4a-11e9-8464-90b11c970001/find_transform-2.log].
> log file: /home/pedram/.ros/log/75cc0b6e-2d4a-11e9-8464-90b11c970001/find_transform-2*.log

Can anyone help?

EpsAvlc, 2019-04-14:
Ｉ write a ros package to convert robosense point cloud into velodyne_cloud.See [here](https://github.com/EpsAvlc/rslidar_to_velodyne)

yulan0215, 2022-09-22:
Hi, do you have any idea about how to use this calibration tool with Ouster LiDAR 128line? 

Do you have any idea about how can I modify the code in the calibration tool to adapt to my Ouster LiDAR? Thx

yulan0215, 2022-09-22:
> I use the lidar of robosense. Only neet to modify some codes.

Hi, I used the Ouster LiDAR with 64 channels. How can I manage to change the code to make this calibration tool to work with my LiDAR, thx!

Qskywalker, 2024-01-18:
Hi, I have the same question. Have you had a solution now?

Is there a common used calibration tool to calibrate Ouster Lidar?

PR #96 (2019-09-06): add hesai 40p support
1. Add parser for /pandar_points provided by [Hesai Lidar Driver](https://github.com/HesaiTechnology/HesaiLidar-ros).
2. Add option in `config_file.txt` to select between Velodyne lidar and Hesai lidar
3. Updated README.md. Add a note about fixing synchronization between Hesai lidar driver and aruco mapping node.

Fixes #42  #48