ankitdhall/lidar_camera_calibration #43: Directory structure for installation.

adioshun, 2018-04-24:
[image]

Hi. I have a question regarding the directory structure for installation. 

This will be the very first question for setup and a stupid. 

In your instruction, 
```
Clone this repository to your machine.
Put the cloned repository, dependencies/aruco_ros and dependencies/aruco_mapping folders in path/to/your/ros/workspace/src and run the following commands,
```

So I did following commands 
```
# Move two dependencies folders under the src
cd ~
git clone https://github.com/ankitdhall/lidar_camera_calibration.git
cd lidar_camera_calibration/
cp -r dependencies/ ./../catkin_ws/src/

# Move entire git folder under the src 
cd ~
cp -r lidar_camera_calibration/ ./catkin_ws/src/
```

Then I type the `catkin_make -DCATKIN_WHITELIST_PACKAGES="aruco_ros;aruco_mapping"` in `~/catkin_ws/`

Is this correct?

karnikram, 2018-04-24:
Hi,
aruco_ros and aruco_mapping should be inside your src directory, not inside the dependencies directory.

Your fourth command should be,
```
cp -r dependencies/aruco_* ../catkin_ws/src/
```

adioshun, 2018-04-24:
Thanks @karnikram. The copied `aruco_ros` folder have multiple subfolders : `aruco`, ` aruco_msgs`, `
 aruco_ros`.

should I keep them in the aruco_ros folder? 
[image]

Or should I move them to upper directory of aruc_ros?

[image]

karnikram, 2018-04-24:
No let the sub folders be as such.