ankitdhall/lidar_camera_calibration #47: find_transform.launch doesn't continue after printing "Reading CameraInfo from topic"

Pratiquea, 2018-05-07:
I am running the launch file on rosbag that I had collected. after a while, it outputs  " Messages of type 0 arrived out of order (will print only once)". The screen with the Aruco markers, where we have to select the corner of the board doesn't even pop up.
I am able to echo the topics and see the lidar and camera data in Rviz as well.

I am using ROS Indigo on Ubuntu  14.04.
[config_file.txt](https://github.com/ankitdhall/lidar_camera_calibration/files/1978906/config_file.txt)
[marker_coordinates.txt](https://github.com/ankitdhall/lidar_camera_calibration/files/1978907/marker_coordinates.txt)
The hardware I am using is VLP_16 and Pointgrey chameleon 3
[image]
[image]

Pratiquea, 2018-05-07:
I followed instructions in #12 and uncommented the ArUco code block in find_transform.h

closed 2018-05-07 by Pratiquea

Surajj4837-th, 2023-09-25:
I think it is find_transform.launch.