ankitdhall/lidar_camera_calibration #48: Cloud window is not visible

sumanthpavuluri, 2018-05-10:
Hi,
Thanks for providing this module for lidar-camera calibration.
The issue I have is when I roslaunch this module I don't see the cloud and polygon windows show up.
I'm not using velodyne lidar but another lidar which has 40 rings as opposed to 16 rings that you use. 
Is there something specific about velodyne lidar that the code assumes?

Please let me know.

Thanks,
Sumanth

karnikram, 2018-05-11:
It works only for Velodyne LiDARs. The package needs ring id information which is provided only by the velodyne drivers.
Check #42

sumanthpavuluri, 2018-05-11:
ring id information is provided by the lidar driver I'm using so I don't think that is the issue. It looks like the callbacks where the lidar points are being loaded are not being called.

sumanthpavuluri, 2018-05-15:
the synchronization between the lidar and camera was off since they were using different time stamps. I can now use this package with a different lidar(hesai) as well.

closed 2018-05-15 by sumanthpavuluri

sangych, 2018-06-26:
@sumanthpavuluri Hello ! Did you do anything else when you use the Hesai-40 lidar? I run this package with velodyne HDL32, and faced the same issue seemed like yours here. I have make the synchronization with "rosbag play --clock ……"

sumanthpavuluri, 2018-06-26:
- Override the lidar PointCloud2.msg header.stamp with ros::TimeStamp in the Hesai lidar's main.cc.
- change the number of rings to 40 in the lidar_camera_calibration package

ruichen-v, 2019-06-11:
See #96. Both point cloud structure and topic timestamp need to be fixed if you use Hesai lidar. Also, change [number of beams](https://github.com/ankitdhall/lidar_camera_calibration/blob/edbcf7823f74515f621aa01d67cfc963b7f32d14/include/lidar_camera_calibration/PreprocessUtils.h#L143) accordingly.