ankitdhall/lidar_camera_calibration #52: how to make 100 iterations?

OYZJ, 2018-06-07:
Hi ankitdhall, May I ask how can I modify your code to run 100 iterations? I met very poor result with only one iteration. And according to your paper it seem the accuracy can be improved a lot by increase the iteration numbers.
Thanks in advance.

blutjens, 2018-06-07:
Change the penultimate line in conf/config_file.txt from "1" to "100". The default parameter is 100, however. Your issue might be caused by something else.

OYZJ, 2018-06-21:
Hello blutjens, Thanks for your help. But the problem is I can't see how this program can reach 100 iteration. I have read through the all the code in src and include. And from the Find_RT.h it seems the iter will never reach to the max_iters without modification.

karnikram, 2018-06-21:
The code does by default run for 100 iterations. The check is done [here](https://github.com/ankitdhall/lidar_camera_calibration/blob/edbcf7823f74515f621aa01d67cfc963b7f32d14/src/Corners.cpp#L255).

Something else is going wrong for you.

lihaolai, 2018-06-26:
hi, I met the same issue.The parameter in the conf/config_file.txt is 100.
[image]
but still only one iteration.
[image]
how can i get it worked on 100 iterations?

sssoubl, 2018-10-11:
repeat "rosbag play --clock ***.bag"!

closed 2019-01-29 by ankitdhall