ankitdhall/lidar_camera_calibration #53: The filtered point cloud was not dispalyed

sangych, 2018-06-11:
I use rosplay to publish the topics needed. the marker was detected by the aruco_mapiing package ,which shown by the window mono8; but the filtered point cloud window was not displayed, I could not mark the line segments. Is anything wrong with my rosbag publishing. How should I publish the topics ? 
Thank You!

karnikram, 2018-06-21:
Make sure all the messages are time synchronized i.e. their timestamps should match.

If you're running them from a bagfile ensure the messages are using the same clock.

sangych, 2018-06-21:
@karnikram thank you for your reply. I have known the usage of the package. Then I stuck in another problem. I change the aruco makers to my own, the aruco_mapping.launch could not detect the marker any more. Would please point out what happened?
/home/sangych/Desktop/3.736.jpg

14212094, 2019-04-02:
@sangych 您好？我是北京其他学校的学生，也在做这个，想问下那个aruco标记是在任何双目相机下都能被检测到吗？这个软件包是当那三个ros节点在list里面就能直接用吗？很抱歉在这里提问 谢谢

closed 2019-09-06 by ankitdhall