ankitdhall/lidar_camera_calibration #54: No response - no windows - no errors

atasoglou, 2018-06-13:
Hello I trying to run the calibration but no window is popping up. Node launches but the last info message is `Reading CameraInfo from configuration file` and then nothing.

Using ROS Indigo with OpenCV 3.4.1 

Any clues? 

Here is the output:
> Size of the frame: 1024 x 512
> Limits: -0.56 to 0.5
> Limits: 2.7 to 3.2
> Limits: -0.99 to -0.43
> Number of markers: 2
> Intensity threshold (between 0.0 and 1.0): 0.0004
> useCameraInfo: 0
> Projection matrix: 
> [551.29785, -0.27212, 501.52975, 0;
>  0, 551.16797, 248.17503, 0;
>  0, 0, 1, 0]
> MAX_ITERS: 10
> [ INFO] [1528900343.889889061]: Reading CameraInfo from configuration file

atasoglou, 2018-06-13:
OK so I run the `aruco_mapper` which - (heads up) - is commented out in the `find_transform.lauch` file.

Now I am getting the following error (crash):
```
Size of the frame: 1024 x 512
Limits: -0.56 to 0.5
Limits: 2.7 to 3.2
Limits: -0.99 to -0.43
Number of markers: 2
Intensity threshold (between 0.0 and 1.0): 0.0004
useCameraInfo: 1
Projection matrix: 
[551.29785, -0.27212, 501.52975, 0;
 0, 551.16797, 248.17503, 0;
 0, 0, 1, 0]
MAX_ITERS: 100
[ INFO] [1528901217.507838434]: Reading CameraInfo from topic
done1
[ WARN] [1528901217.766896234]: Messages of type 0 arrived out of order (will print only once)
[ INFO] [1528901218.078632845]: Camera info received at 1.5289e+09
[ INFO] [1528901218.078676401]: Velodyne scan received at 1.5289e+09
[ INFO] [1528901218.078692674]: marker_6dof received at 1.5289e+09
[find_transform-1] process has died [pid 1566, exit code -11, cmd /home/tanasaki/ROS/catkin_ws/devel/lib/lidar_camera_calibration/find_transform __name:=find_transform __log:=/home/tanasaki/.ros/log/62a69876-6f11-11e8-9a69-6045cb6d5ae1/find_transform-1.log].
log file: /home/tanasaki/.ros/log/62a69876-6f11-11e8-9a69-6045cb6d5ae1/find_transform-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
```

The aruco detection seems to be fine.

karnikram, 2018-06-21:
It seems like your messages are out of sync. The timestamps of all the messages must match and must be using the same clock, sim clock if you're using a bagfile.

closed 2018-06-29 by ankitdhall

vllsm, 2019-09-05:
hi, atasoglou, I am confronting the same error as you. "[find_transform-1] process has died: So could you please tell me your solution? thank you so much in advance

vllsm, 2019-09-05:
@atasoglou 
@ankitdhall 
please

xuhangxx, 2020-05-16:
i delete commas in the  .ini  file, and solve the same problem.