ankitdhall/lidar_camera_calibration #55: marker_coordinate confused.

ZhengPan2014, 2018-06-16:
I'm confused about the coordination description of aruco maker as you stress the importance of Y axis pointing outward while XY planer is lying on marker planer like the following picture in our daily usage. X(red), Y(green), Z(blue). Do that mean any special function of this difference? thank you
[image]

karnikram, 2018-06-21:
The convention is to ensure that the X and Z axes are aligned along the edges of the cardboard marker. This makes it easier to estimate the corner locations using the board dimensions as it is done [here](https://github.com/ankitdhall/lidar_camera_calibration/blob/edbcf7823f74515f621aa01d67cfc963b7f32d14/include/lidar_camera_calibration/Find_RT.h#L273).

Regarding your second question, ensure that the topic name, intrinsic calibration values, and marker dimensions are set properly.

ZhengPan2014, 2018-06-21:
I run aruco_mapping package independently, and no marker detect while all seems good. the runtime shotcut is as follow:
[image]
any advice? thanks for your reply.

karnikram, 2018-06-21:
Maybe you could vary the distance between the markers and your camera and try. Also ensure that the marker size in your launch file is correct.

ZhengPan2014, 2018-06-22:
@karnikram yes, I did vary the distance to test aruco_mapping package independently. unfortunately, nothing new happens, shotcut is as follow:
[image]

one question, Should the dictionary parameter of markers be passed explicitly? as showed above, I generated makers ID 512 and 513 with 7 dictionary parameter whose detail is available [here](http://wiki.ros.org/aruco_detect).

by the way, aruco_detect package works well for the same scene. the runtime shotcut is as follow:
[image]

looking forward to your reply.

karnikram, 2018-06-22:
This seems very weird. Are you *sure* the values in the .ini file are
right? The same values that are published in the camera_info topic? And no,
you don't have to pass any other parameter.

ZhengPan2014, 2018-06-25:
@karnikram yes, after running aruco_detect package successfully with topic **/mynteye/left/image_rect/compressed** and **/mynteye/left/camera_info**, I copy **/mynteye/left/camera_info** correspondding paramters to **zed_left_uurmi.ini** file and run aruco_mapping with **/mynteye/left/image_rect** and no maker detected.

XiaoJianFang, 2018-06-28:
I annotated this sentence in aruco_mapping.cpp

 lidar_camera_calibration-master/dependencies/aruco_mapping/src/aruco_mapping.cpp

440   ros::Duration(BROADCAST_WAIT_INTERVAL).sleep();

ericzzj1989, 2019-10-17:
@ZhengPan2014 Hi, have you been able to detect the markers? I trying the package and also unable to detect anything

narutojxl, 2019-10-28:
Hi @ZhengPan2014,  the same problem with you, aruco_mapping can not detect marker, but aruco_detect with correct dictionary argument can detect the above marker. Aruco_mapping doesn't have this param. BTW, did you have solved this problem? Could we have a discussion, thanks a lot!

ericzzj1989, 2019-10-28:
Hello，

Sure. My wechat is ericzzj. Thanks.

Best,
Zhenjun

| |
ericzzj
|
|
邮箱：ericzzj@163.com
|

Signature is customized by Netease Mail Master

On 10/28/2019 20:34, narutojxl wrote:

Hi @ZhengPan2014, the same problem with you, aruco_mapping can not detect marker, but aruco_detect with correct dictionary argument can detect the above marker. Aruco_mapping doesn't have this param. BTW, did you have solved this problem? Could we have a discussion, thanks a lot!

narutojxl, 2019-11-04:
Hi @ericzzj1989, I figure out this problem.  The aruco_mapping and author modified aruco_mapping can only detect original aruco marker type, [aruco_detect](http://wiki.ros.org/aruco_detect) can only detect dictionary aruco marker type, see [online aruco marker generator](http://chev.me/arucogen/). When using this web site generate marker, the set "Marker size, mm" does not match the actual marker size, you must manual measure with a ruler.

ericzzj1989, 2019-11-04:
@narutojxl Many thanks. In addition, I encounter an another problem.
I am currently doing indoor UAV localization using multiple Aruco marker information, but I encounter a problem about the pose calculation. 

The following is my setup:
Use aruco_mapping software in github;
Use aruco software in github;
Print and use the original aruco markers;
Set the necessary parameters (marker numbers, size, etc).

The problem is that the calculated poses are not stable and even sometimes jump to the wrong data. Did you encounter the same problem. Could we have a discussion, thanks a lot!

narutojxl, 2019-11-04:
Hi @ericzzj1989, i only use this open source for lidar camera calibration, treating aruco_mapping as a black box and do not read its source code. Maybe i only have two markers,  the calculated markers pose are stable.

ericzzj1989, 2019-11-04:
@narutojxl Ok. Many thanks!