ankitdhall/lidar_camera_calibration #56: aruco_mapping don't detect the marker

sangych, 2018-06-21:
@karnikram I thought I have known the usage of the package. Then I stuck in another problem. I change the aruco makers to my own, the aruco_mapping.launch could not detect the marker any more. Would please point out what happened? Is there any requires about the marker selection or any problems with the method I setup the marker boards ? The id of the two markers are 16 and 48 here
[image]

sangych, 2018-06-21:
What's your mean of the dimensions? When I just replace the image by other one's setup pics, regardless the intrinsic params, the markers could be successfully detected , it makes me really confused

karnikram, 2018-06-21:
I mean the marker_size parameter, although it might not make too much of a
difference.

karnikram, 2018-06-21:
I've never really faced this issue, it has always worked for me after setting the right parameters. Maybe you could vary the distance between the markers and your camera and try.

sangych, 2018-06-21:
OK,really thanks for your reply, I'll post here if I found the issue. But one more question, could I believe that the id of the aruco markers and the direction of the boards  don't matter?

karnikram, 2018-06-21:
They will not affect the detection. But they will affect the calibration,
read the README for that.

sangych, 2018-06-24:
I'am so sorry that I have not found the issue yet. I plan to change my marker board. In order to keep things the same , would please send me the marker images (id 26 and 582) to me? I was doubt that the my marker pics from net was not correct. Thank you! My email address : sangych@foxmail.com

karnikram, 2018-06-24:
Not sure if that'll change anything, but you can find an online marker
generator. Like this - http://chev.me/arucogen/

sangych, 2018-06-24:
I just don't know that if the aruco image is the same for every id. The image of markers are different from the generator you give me, which have the same id.

sangych, 2018-06-25:
@karnikram Hi ! It was really caused by the wrong aruco pics that the aruco_mapping didn't work. The aruco_mapping detect the markers, but the find_transform progress exit with error code -11
it seem like this:
……
[ WARN] [1529931891.588262623]: Wrong calibration data, check calibration file and filepath
[ INFO] [1529931891.592535929]: Reading CameraInfo from configuration file
26=(299.39,357.62) (420.977,422.573) (358.899,544.87) (239.84,479.634) Txyz=-1.21141 0.330626 5.03774 Rxyz=-1.04231 -1.7179 1.45041 
582=(677.846,375.878) (794.966,440.566) (729.804,561.661) (612.02,497.12) Txyz=0.143135 0.398753 5.24757 Rxyz=-1.05146 -1.67171 1.52541 
[ INFO] [1529931892.058793679, 1529912142.884315556]: First marker with ID: 26 detected
[ WARN] [1529931892.164194277, 1529912142.995289563]: TF to MSG: Quaternion Not Properly Normalized
[ INFO] [1529931892.166864302, 1529912142.995289563]: Velodyne scan received at 1.52991e+09
[ INFO] [1529931892.166947729, 1529912142.995289563]: marker_6dof received at 1.52991e+09

Initial Rot 0.000796274    -0.999999 -0.000796274
           0  0.000796274           -1
           1  0.000796274  6.34053e-07
26=(299.39,357.62) (420.977,422.573) (358.899,544.87) (239.84,479.634) Txyz=-1.21141 0.330626 5.03774 Rxyz=-1.04231 -1.7179 1.45041 
582=(677.846,375.878) (794.966,440.566) (729.804,561.661) (612.02,497.12) Txyz=0.143135 0.398753 5.24757 Rxyz=-1.05146 -1.67171 1.52541 
[find_transform-2] process has died [pid 20985, exit code -11, cmd /home/sangych/catkin/devel/lib/lidar_camera_calibration/find_transform __name:=find_transform __log:=/home/sangych/.ros/log/d744d964-7870-11e8-a974-9cb6d0fb9f17/find_transform-2.log].
log file: /home/sangych/.ros/log/d744d964-7870-11e8-a974-9cb6d0fb9f17/find_transform-2*.log
26=(299.39,357.62) (420.977,422.573) (358.899,544.87) (239.84,479.634) Txyz=-1.21141 0.330626 5.03774 Rxyz=-1.04231 -1.7179 1.45041 
582=(677.846,375.878) (794.966,440.566) (729.804,561.661) (612.02,497.12) Txyz=0.143135 0.398753 
……

And I found the master.log shown:
……
[rosmaster.master][INFO] 2018-06-25 21:04:55,656: publisherUpdate[/tf] -> http://sangych-ubuntu16:42377/ []: sec=0.00, exception=[Errno 111] Connection refused
[rosmaster.threadpool][ERROR] 2018-06-25 21:04:55,657: Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosmaster/threadpool.py", line 218, in run
    result = cmd(*args)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosmaster/master_api.py", line 210, in publisher_update_task
    ret = xmlrpcapi(api).publisherUpdate('/master', topic, pub_uris)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1243, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1602, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1283, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1311, in single_request
    self.send_content(h, request_body)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1459, in send_content
    connection.endheaders(request_body)
  File "/usr/lib/python2.7/httplib.py", line 1053, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python2.7/httplib.py", line 897, in _send_output
    self.send(msg)
  File "/usr/lib/python2.7/httplib.py", line 859, in send
    self.connect()
  File "/usr/lib/python2.7/httplib.py", line 836, in connect
    self.timeout, self.source_address)
  File "/usr/lib/python2.7/socket.py", line 575, in create_connection
    raise err
error: [Errno 111] Connection refused
  ……

Could you please find what's the issue ?

karnikram, 2018-06-26:
Seems like an error at the level of your ROS master. Probably has nothing to do with this package.

sangych, 2018-06-26:
The lidar I used is velodyne HDL32 , was there anything changed?

villanuevab, 2018-07-25:
@sangych you need to change the number of rings in PreprocessUtils.h if your lidar has != 16 rings (the default). The line you have to change is: https://github.com/ankitdhall/lidar_camera_calibration/blob/master/include/lidar_camera_calibration/PreprocessUtils.h#L143

closed 2019-01-29 by ankitdhall

roylu307, 2020-03-05:
hello @villanuevab , how could I change the number of ring from that line of the code you were pointing to?