ankitdhall/lidar_camera_calibration #57: how to examine the result with a monocular camera a velodyne16?

lihaolai, 2018-06-26:
Hi ankitdhall, i have got a result from lidar_camera_calibration with a velodyne16 and a monocular camera. But i don't have a zed camera. So how can i examine the result with the monocular camera and velodyne16?

karnikram, 2018-06-26:
#19 could help.

lihaolai, 2018-06-29:
than you

GeeKmyself, 2018-09-27:
@lihaolai , hi , could u tell me how to  calibrate lidar and mono camera by this method , i am confused with how to find "3D Point correspondences in the Camera Frame" in paper.

lihaolai, 2018-11-28:
@Chrislzy1993 i used depth pointclout from rgbd camera to correspond to the 3d velodyne16 pointcloud. but it is not very good in result. maybe we can add qq or wechat to discuss this together

closed 2019-09-06 by ankitdhall