ankitdhall/lidar_camera_calibration #58: Calibration result is not accurate!

topcomma, 2018-07-25:
Hi Ankidhall,

Firstly thanks for your great work.

I have tried to use the function to calibrated my lidar and camera on car, Velodyne VLP16 LiDAR on car and camera is the front camera on the underground garage.
I manually measured the correspdondence between camera and lidar is about xyz(0.05,0.37,0.9)  and unit is meter.

But now I tried many times using your code, the result is not very well.

Number of points: 8
     1.66108    0.0107726    0.0114646
    -0.24578      0.51019  -0.00319998
   -0.050756    0.0132938 -2.61004e-05
Rotation matrix: 
  0.992472   0.117913  0.0331166
 -0.116926   0.992678 -0.0303074
-0.0364477   0.026207   0.998992
Rotation in Euler angles: 
173.294
177.924
-178.51
Translation: 
-0.0688622
 -0.376567
 -0.656073
RMSE: 0.0574024
Rigid-body transformation: 
  0.992472   0.117913  0.0331166 -0.0688622
 -0.116926   0.992678 -0.0303074  -0.376567
-0.0364477   0.026207   0.998992  -0.656073
         0          0          0          1
--------------------------------------------------------------------
After 100 iterations
--------------------------------------------------------------------
Average translation is:
-0.0570178
 -0.350666
 -0.660128
Average rotation is:
  0.992761   0.116792   0.028013
 -0.115593   0.992443 -0.0411774
-0.0326105  0.0376412   0.998759
Average transformation is: 
  0.992761   0.116792   0.028013 -0.0570178
 -0.115593   0.992443 -0.0411774  -0.350666
-0.0326105  0.0376412   0.998759  -0.660128
         0          0          0          1
Final rotation is:
 0.0288035  -0.992645  -0.117583
-0.0412694    0.11635   -0.99235
  0.998733  0.0334358 -0.0376146
Final ypr is:
  2.18014
 -1.62114
-0.726651
Average RMSE is: 0.0644712
RMSE on average transformation is: 0.0578473

I have attached the config files, please help me out .

Thanks in advance.

[config_file.txt](https://github.com/ankitdhall/lidar_camera_calibration/files/2226218/config_file.txt)

[marker_coordinates.txt](https://github.com/ankitdhall/lidar_camera_calibration/files/2226219/marker_coordinates.txt)

[gmsl_camera.txt](https://github.com/ankitdhall/lidar_camera_calibration/files/2226220/gmsl_camera.txt)
Xingui

topcomma, 2018-07-25:
I do not know how to more conveniently upload the corresponding ROS bag.

xinwf, 2018-10-25:
Your camera's parameter is wrong, height is 1280, not 1208.

closed 2019-01-29 by ankitdhall