ankitdhall/lidar_camera_calibration #60: how to obtain 3D points in the camera frame

xccoming, 2018-08-06:
how to obtain 3D points in the camera frame, for mono camera,  in real scale ?
in paper:
"The tags provide [R|t] between the camera and the center of the marker.
This transform can be used to convert corner points from the marker’s frame-ofreference
(which is the cardboard plane with the origin being the center of the
ArUcO marker) to the camera’s frame-of-reference. This allows to obtain the
corners as 3D points in the camera frame. "
but this t is not in real scale!

GeeKmyself, 2018-09-27:
@xccoming , i am confused too,  did u solve this problem.

icameling, 2018-09-28:
从aruco_mapping 得到tag到camera的转换矩阵 (R|t) ，这里单位是米
由 /conf/marker_coordinates.txt 输入标定板尺寸，得到标定板四个角点在tag坐标系下的位置，单位也是米
将在tag坐标系下的四个角点经Rt转换到相机坐标系，得到在相机坐标系下4个角点的空间3D位置。
两个标定板就是8个点。

对于激光则用直线拟合的方法得到这8个角点位置，问题变成了： 已知两组对应的8D点，求两组点之间的转换关系。用Least-Squares Rigid Motion Using SVD 论文里面提出方法即可解决，也就是本论文里面的4.1部分。

icameling, 2018-09-28:
从aruco_mapping 得到tag到camera的转换矩阵 (R|t) ，这里单位是米
由 /conf/marker_coordinates.txt 输入标定板尺寸，得到标定板四个角点在tag坐标系下的位置，单位也是米
将在tag坐标系下的四个角点经Rt转换到相机坐标系，得到在相机坐标系下4个角点的空间3D位置。
两个标定板就是8个点。

对于激光则用直线拟合的方法得到这8个角点位置。

于是问题变成了： 已知两组对应的8D点，求两组点之间的转换关系。用Least-Squares Rigid Motion Using SVD 论文里面提出方法即可解决，也就是本论文里面的4.1部分。

GeeKmyself, 2018-09-29:
@icameling , 谢谢你， 请问下对于marker和tag的尺寸应该怎样确定？

xinwf, 2018-10-25:
@icameling  我运行这个之后卡在了回调函数前面，怎么回事呢，输出"done1“之后就没有任何输出了，也没有弹出窗口，topic都是有数据的。

narutojxl, 2019-11-01:
Hi @icameling, how to get 3D points corners of calibration board in laser frame, details see [#107](https://github.com/ankitdhall/lidar_camera_calibration/issues/107), thanks a lot!

SinghiDivyanshu, 2020-01-27:
hi @xccoming ,
Did you solve it. I'm not able to obtain 3D points in the camera frame, for mono camera.