ankitdhall/lidar_camera_calibration #64: How to get accurate inital ros_x, rot_y, rot_z?

oaix, 2018-10-26:
The precision of inital ros_x, rot_y, rot_z will impact final rotation matrix.
```
qlidarToCamera = Eigen::AngleAxisd(config.initialRot[2], Eigen::Vector3d::UnitZ()) *
                   Eigen::AngleAxisd(config.initialRot[1], Eigen::Vector3d::UnitY()) *
                   Eigen::AngleAxisd(config.initialRot[0], Eigen::Vector3d::UnitX());
lidarToCamera = qlidarToCamera.matrix();
Eigen::Matrix3d final_rotation = rotation_avg * lidarToCamera;
```

xinwf, 2018-10-27:
Can we have a communication about this project? I can't get any result, when the window pop up, there is no any dots on it, just a black window, I don't know what should I do

oaix, 2018-10-29:
+ First of all, you should make sure your point cloud on cardboard can be seen in the view of camera, which is impacted by the initial  ros_x, rot_y and rot_z.
+ Second, edge points of cardboard should be detected. If it is successful, it will show in `cloud` window as below.
[image]
+ Then you can  draw a quadrilateral around the line being marked. Click and press a key (any key you like) to confirm the corner of the quadrilateral. In the `polygon` window you will see as below. 
[image]

xinwf, 2018-10-30:
@oaix, 说错了，高度差大概有80cm

oaix, 2018-10-30:
@xinwf 你给一个平移量就可以了，源码里面给的[0, 0, 0]，也就是下面的x, y, z的值
transform(board_edge_cloud, x, y, z, config.initialRot[0], config.initialRot[1], config.initialRot[2]);

closed 2018-11-01 by oaix

ailovejinx, 2024-04-22:
> @oaix ,非常感谢！不过再多问一些啊，这里的x，y，z应该怎么样取值呢，首先，这个校准是激光雷达到相机，所以应该以相机为原点，xyz是雷达到相机的平移量，对吧？第二个就是，xyz取值问题，是距离量还是坐标量，距离量就都是正的，坐标量就会有正负，如果是坐标量的话，是激光雷达在相机坐标系中的坐标吗，还是坐标取反表示平移量，例如，雷达在相机坐标系（Z向前，Y向下，X向右）中的坐标为（0，-0.8，-0.5），那么平移量就是[0, -0.8, -0.5]还是[0, 0.8, 0.5]

hi, xinwf, I also met this problem. Have you figured out the problem?