ankitdhall/lidar_camera_calibration #67: preciseness of final rotation depends on accurate initial rotation?

oaix, 2018-11-01:
@karnikram @jediofgever @villanuevab
Thank you for your reply timely and patiently.
In consideration of `Eigen::Matrix3d final_rotation = rotation_avg * lidarToCamera`[#30](https://github.com/ankitdhall/lidar_camera_calibration/issues/30) and `lidarToCamera` is the initial configurable rotation between the lidar and the camera. The accuracy of initial `rot_x`, `rot_y`, and `rot_z` will impact the preciseness of final rotation. How can I get an accurate initial rotation to guarantee the preciseness of final rotation?
Here is my calibration result. Transform point cloud of lidar to camera coordinate. The color of point represents the depth of point in lidar coordinate. Could you please give me some advise to improve calibration preciseness? Thank you very much!
[image]

oaix, 2018-11-01:
I got the problem. Thanks all the same!

closed 2018-11-01 by oaix

SiddhantNadkarni, 2020-08-07:
How did you resolve this?

SiddhantNadkarni, 2020-08-07:
I have the same issue..

jamesheatonrdm, 2021-11-03:
> I got the problem. Thanks all the same!

Please tell us how you fixed the problem.