ankitdhall/lidar_camera_calibration #72: Calibration error: relative rotation between the lidar and camera is off

sunemily, 2018-12-11:
Hi, 

We've been using your package to calibrate between a Velodyne VLP16 LiDAR and a ZED camera. We were able to get a relatively good result but our sample Velodyne to ZED projection results seem to be shifted to the left. It seems like the relative rotation between the lidar and camera is off, and were wondering it possible there’s an axis/sign error with the yaw rotation in the transform? 

I've attached the config files used in the calibration as well as some sample project result images with LiDAR points shown to be shifted to the left.

Are these types of slight calibration errors expected? If not, what can we do to obtain a better calibration? Thank you!

--------------------------------------------------------------------
After 100 iterations
--------------------------------------------------------------------
Average translation is:
0.0788898
-0.130423
-0.245756
Average rotation is:
   0.997837  -0.0464536  -0.0465055
  0.0465616    0.998915  0.00124173
  0.0463973 -0.00340441    0.998917
Average transformation is: 
   0.997837  -0.0464536  -0.0465055   0.0788898
  0.0465616    0.998915  0.00124173   -0.130423
  0.0463973 -0.00340441    0.998917   -0.245756
          0           0           0           1
Final rotation is:
-0.0457109  -0.997911   0.045659
0.00127881 -0.0457652  -0.998951
  0.998954 -0.0456046  0.0033681
Final ypr is:
 3.11362
-1.52505
-1.49708
Average RMSE is: 0.0429108
RMSE on average transformation is: 0.0668673

[config_file.txt](https://github.com/ankitdhall/lidar_camera_calibration/files/2665741/config_file.txt)
[marker_coordinates.txt](https://github.com/ankitdhall/lidar_camera_calibration/files/2665742/marker_coordinates.txt)
[zed_left_uurmi.txt](https://github.com/ankitdhall/lidar_camera_calibration/files/2665743/zed_left_uurmi.txt)

[image]
[image]
[image]

hunkyu, 2019-02-19:
@sunemily Can you solve it?

chenwinki, 2019-05-07:
I have the same problem with you, 

**--------------------------------------------------------------------
After 100 iterations
--------------------------------------------------------------------**
Average translation is:
-0.136269
0.0661312
-0.128244
Average rotation is:
  0.960093  0.0247276  -0.278585
-0.0430343   0.997283 -0.0597897
  0.276349  0.0693924   0.958549
Average rotation is:
 0.0622946
  -0.28232
-0.0257497
Average transformation is: 
  0.960093  0.0247276  -0.278585  -0.136269
-0.0430343   0.997283 -0.0597897  0.0661312
  0.276349  0.0693924   0.958549  -0.128244
         0          0          0          1
Final rotation is:
-0.0814494  -0.996602 -0.0122436
-0.0645887  0.0175365  -0.997758
  0.994582  -0.080476 -0.0657976
Final ypr is:
   1.63665
-0.0122439
   1.65234
Average RMSE is: 0.0447147
RMSE on average transformation is: 0.0484879

The RMSE is small, but the project result is not good.
[image]

Andyyyyyyyyyyyy, 2019-08-09:
> I have the same problem with you,
>
> **-------------------------------------------------------------------- After 100 iterations --------------------------------------------------------------------** [...]
I have got the rigid-body transformation using this pkg, but I don't how to visualize the  lidar-to-camera projection results, can you show me how to project the point cloud to the camera? Any advice is appreciated!

SinghiDivyanshu, 2020-02-12:
hi @Andyyyyyyyyyyyy 

Are you able to visualize the LIDAR points on camera points. can you help, i'm also got stuck after getting rigid-body transformation.

Andyyyyyyyyyyyy, 2020-02-16:
may be this would help youSee fusion.py------------------ Original ------------------From: SinghiDivyanshu  <notifications@github.com>Date: Wed,Feb 12,2020 3:23 PMTo: ankitdhall/lidar_camera_calibration <lidar_camera_calibration@noreply.github.com>Cc: Andyyyyyyyyyyyy <13002573282@163.com>, Mention <mention@noreply.github.com>Subject: Re: [ankitdhall/lidar_camera_calibration] Calibration error: relativerotation between the lidar and camera is off (#72)hi @Andyyyyyyyyyyyy
Are you able to visualize the LIDAR points on camera points. can you help, i'm also got stuck after getting rigid-body transformation.

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/ankitdhall/lidar_camera_calibration/issues/72?email_source=notifications\u0026email_token=AL2IQDVLG72OT4LRMWZFR5LRCOPYPA5CNFSM4GJSHSCKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOELPW3QY#issuecomment-585067971",
"url": "https://github.com/ankitdhall/lidar_camera_calibration/issues/72?email_source=notifications\u0026email_token=AL2IQDVLG72OT4LRMWZFR5LRCOPYPA5CNFSM4GJSHSCKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOELPW3QY#issuecomment-585067971",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

SiddhantNadkarni, 2020-08-07:
I have the same issue. Was anyone able to resolve this?

AnthonyLadewig, 2020-11-02:
Hi @sunemily 

How did you project the lidar data onto an image?

Thanks!