ankitdhall/lidar_camera_calibration #75: Calibration result is not good

zheng-rong, 2019-01-07:
Hi all,
I am trying to calibrate a VLP-16 and monocular camera (equipped with a lens with distortion).
I followed the steps in the wiki, but the result seems not reasonable.
The following picture shows my sensor configuration:
[image]

and the result is:
[image]
[image]

By far I haven't figure out the reason for this bad calibration result.

I am wondering if this is caused by the camera intrinsic matrix. I am using a monocular camera with distortion, and I got the K(intrinsic) and D(distortion) by Kalibr, and undistort the images by openCV function. IN the aruco_mapping config file, I filled in the K and D as required, and feed it with mono/image_raw(distorted). But in lidar_camera_calibration, only K is specified. So I use getOptimalCameraMatrix in openCV to get a new_K (I think this new_K is the intrinsic for undistorted camera image), and feed the new_K and mono/image_rect to the lidar_camera_calibration. Is this correct?

Any help will be appreciated. The config files I am using are attached.

Thanks in advance,
Zheng

[vi_0003.txt](https://github.com/ankitdhall/lidar_camera_calibration/files/2731717/vi_0003.txt)
[config_file.txt](https://github.com/ankitdhall/lidar_camera_calibration/files/2731721/config_file.txt)
[lidar_camera_calibration.txt](https://github.com/ankitdhall/lidar_camera_calibration/files/2731726/lidar_camera_calibration.txt)
[marker_coordinates.txt](https://github.com/ankitdhall/lidar_camera_calibration/files/2731727/marker_coordinates.txt)

zheng-rong, 2019-01-08:
Emmm... I got the wrong marker size in the config file. After revising that, the calibration result looks good. The RMSE is 0.0327.
But I still have the question: Does this calibration utility can directly deal with the camera with distortion?

hunkyu, 2019-02-15:
Can you send your modified configuration file?

zheng-rong, 2019-02-17:
Please see the attached file.
Thanks.
[marker_coordinates_1.txt](https://github.com/ankitdhall/lidar_camera_calibration/files/2872622/marker_coordinates_1.txt)

hunkyu, 2019-02-19:
[image]
I don't know why it is projected?

hunkyu, 2019-03-01:
@zheng-rong Can you leave a mailbox and follow up? My email 513459395@qq.com

zlbing, 2019-03-29:
@zheng-rong hi,
could you tell me how to make the RMSE small, i used you dataset, and can't get right result

chenwinki, 2019-04-25:
@hunkyu Could you give me the pdf or picture of markers? Thank you!

Elkulas, 2019-05-12:
@hunkyu how to project points to image?

SinghiDivyanshu, 2020-01-24:
hi @Elkulas @hunkyu @zheng-rong 
can you able to project points to image.....i'm stuck after finding rigid body transformation

wenbuming, 2024-08-28:
hi, have you figured out which one should be used, distort or undistort?