ankitdhall/lidar_camera_calibration #80: [find_transform-2] process has died

pedghz, 2019-02-10:
The prcess get killed regurarly! I have the problem that the find_transform process got killed, not always, but most of the times! aproximately 9 times out of 10 times of runing the launch file, the process got killed. What is going worng here? Here I attach the output from running the launch file:

pedram@WS-4022-NPC:~$ roslaunch lidar_camera_calibration find_transform.launch
... logging to /home/pedram/.ros/log/75cc0b6e-2d4a-11e9-8464-90b11c970001/roslaunch-WS-4022-NPC-31673.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://WS-4022-NPC:44503/

SUMMARY
========

PARAMETERS
 * /aruco_mapping/calibration_file: /home/pedram/catk...
 * /aruco_mapping/marker_size: 0.212
 * /aruco_mapping/num_of_markers: 2
 * /aruco_mapping/roi_allowed: False
 * /aruco_mapping/space_type: plane
 * /lidar_camera_calibration/camera_frame_topic: /image_raw_sync
 * /lidar_camera_calibration/camera_info_topic: /camera_info_sync
 * /lidar_camera_calibration/velodyne_topic: /rslidar_points_sync
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    aruco_mapping (aruco_mapping/aruco_mapping)
    find_transform (lidar_camera_calibration/find_transform)

ROS_MASTER_URI=http://localhost:11311

process[aruco_mapping-1]: started with pid [31696]
process[find_transform-2]: started with pid [31697]
[ INFO] [1549820982.143502743]: Calibration file path: /home/pedram/catkin-ws/src/aruco_mapping/data/zed_left_uurmi.ini
[ INFO] [1549820982.143570207]: Number of markers: 2
[ INFO] [1549820982.143607660]: Marker Size: 0.212
[ INFO] [1549820982.143631209]: Type of space: plane
[ INFO] [1549820982.143656830]: ROI allowed: 0
[ INFO] [1549820982.143675896]: ROI x-coor: 32767
[ INFO] [1549820982.143695376]: ROI y-coor: 32767
[ INFO] [1549820982.143714975]: ROI width: 0
[ INFO] [1549820982.143733998]: ROI height: 4280747
[ INFO] [1549820982.145394225]: Calibration data loaded successfully
Size of the frame: 800 x 450
Limits: -6 to 6
Limits: -7 to 7
Limits: -2 to 2
Number of markers: 2
Intensity threshold (between 0.0 and 1.0): 0.05
useCameraInfo: 1
Projection matrix: 
[1421.1016, 0, 389.20328, 0;
 0, 1423.6897, 222.87619, 0;
 0, 0, 1, 0]
MAX_ITERS: 20
[ INFO] [1549820982.156012321]: Reading CameraInfo from topic
done1
26=(218.184,240.842) (149.744,304.732) (83.1856,236.119) (151.451,170.895) Txyz=-0.527223 0.0342224 3.12697 Rxyz=-0.661288 1.82979 -1.57776 
582=(582.888,258.202) (509.872,319.664) (446.398,248.201) (518.841,185.685) Txyz=0.274132 0.0663473 3.10322 Rxyz=-0.817062 1.93256 -1.51535 
[ INFO] [1549820982.551370550]: First marker with ID: 26 detected
[ WARN] [1549820982.554996500]: TF to MSG: Quaternion Not Properly Normalized
26=(218.139,240.921) (149.724,304.714) (83.1935,236.204) (151.503,170.897) Txyz=-0.527472 0.0343165 3.12845 Rxyz=-0.660293 1.82778 -1.579 
582=(582.908,258.234) (509.838,319.67) (446.463,248.229) (518.822,185.687) Txyz=0.274221 0.0663967 3.10406 Rxyz=-0.818842 1.93217 -1.51661 
26=(218.174,240.854) (149.78,304.654) (83.1606,236.131) (151.512,170.936) Txyz=-0.527373 0.0342395 3.12815 Rxyz=-0.656092 1.82903 -1.57443 
582=(582.913,258.23) (509.873,319.64) (446.452,248.192) (518.835,185.679) Txyz=0.274194 0.0663483 3.10355 Rxyz=-0.819217 1.93401 -1.51476 
[ INFO] [1549820982.638262530]: Camera info received at 1.54982e+09
[ INFO] [1549820982.638318152]: Velodyne scan received at 1.54982e+09
[ INFO] [1549820982.638360950]: marker_6dof received at 1.54982e+09
Failed to find match for field 'ring'.
[... 97 lines cut ...]
26=(218.203,240.833) (149.761,304.679) (83.2324,236.094) (151.5,170.887) Txyz=-0.5273 0.0341804 3.12787 Rxyz=-0.660634 1.82938 -1.57739 
582=(582.952,258.197) (509.873,319.64) (446.459,248.2) (518.851,185.679) Txyz=0.274197 0.066328 3.10313 Rxyz=-0.817106 1.93329 -1.51456 
26=(218.181,240.863) (149.747,304.711) (83.2481,236.136) (151.49,170.853) Txyz=-0.527309 0.0342098 3.12779 Rxyz=-0.663309 1.82912 -1.57988 
582=(582.975,258.232) (509.83,319.649) (446.488,248.174) (518.843,185.686) Txyz=0.27419 0.0663394 3.1031 Rxyz=-0.81912 1.93392 -1.51436 
26=(218.186,240.869) (149.775,304.703) (83.2026,236.198) (151.489,170.879) Txyz=-0.527294 0.0342637 3.12771 Rxyz=-0.659604 1.82867 -1.5782 
582=(582.97,258.236) (509.859,319.624) (446.459,248.2) (518.832,185.695) Txyz=0.274126 0.0663424 3.10248 Rxyz=-0.81763 1.93568 -1.51206 
26=(218.16,240.865) (149.779,304.699) (83.2029,236.118) (151.518,170.887) Txyz=-0.527322 0.0342212 3.12788 Rxyz=-0.660943 1.82919 -1.57781 
582=(582.965,258.179) (509.903,319.594) (446.41,248.228) (518.828,185.685) Txyz=0.274065 0.0662977 3.10179 Rxyz=-0.814087 1.93602 -1.51064 
26=(218.171,240.864) (149.802,304.697) (83.1638,236.131) (151.524,170.892) Txyz=-0.527217 0.0342287 3.12729 Rxyz=-0.658854 1.82925 -1.57633 
582=(582.987,258.222) (509.888,319.6) (446.422,248.186) (518.841,185.707) Txyz=0.274039 0.066307 3.10137 Rxyz=-0.815813 1.93757 -1.50916 
26=(218.164,240.869) (149.746,304.693) (83.1545,236.12) (151.504,170.873) Txyz=-0.527253 0.0342096 3.1272 Rxyz=-0.659371 1.82883 -1.57688 
582=(582.97,258.236) (509.871,319.614) (446.447,248.243) (518.823,185.686) Txyz=0.274102 0.0663525 3.10222 Rxyz=-0.816265 1.93578 -1.51163 
26=(218.202,240.834) (149.734,304.704) (83.2438,236.128) (151.507,170.922) Txyz=-0.527466 0.0342381 3.12885 Rxyz=-0.659807 1.82747 -1.57915 
582=(582.909,258.185) (509.885,319.653) (446.455,248.196) (518.866,185.661) Txyz=0.27427 0.0663277 3.10405 Rxyz=-0.817735 1.9312 -1.51717 
26=(218.226,240.804) (149.8,304.679) (83.1927,236.103) (151.465,170.963) Txyz=-0.527243 0.0342203 3.12754 Rxyz=-0.657071 1.83055 -1.57396 
582=(582.904,258.189) (509.89,319.649) (446.414,248.225) (518.851,185.665) Txyz=0.274171 0.0663344 3.10325 Rxyz=-0.816127 1.93213 -1.5156 
---------Moving on to next marker--------
26=(218.118,240.899) (149.78,304.688) (83.177,236.223) (151.507,170.873) Txyz=-0.527463 0.034289 3.12849 Rxyz=-0.659493 1.8281 -1.57869 
582=(582.921,258.252) (509.894,319.676) (446.449,248.24) (518.87,185.675) Txyz=0.274214 0.0663969 3.10334 Rxyz=-0.81818 1.93246 -1.51595 
26=(218.174,240.854) (149.756,304.676) (83.2199,236.133) (151.505,170.868) Txyz=-0.527373 0.0342026 3.12817 Rxyz=-0.660616 1.82871 -1.57829 
582=(582.946,258.224) (509.871,319.648) (446.473,248.174) (518.845,185.67) Txyz=0.274205 0.0663281 3.10331 Rxyz=-0.819841 1.93393 -1.51504 
26=(218.17,240.853) (149.725,304.694) (83.2834,236.124) (151.479,170.869) Txyz=-0.527483 0.0342056 3.12878 Rxyz=-0.664039 1.82899 -1.5805

I also attach a picture from the time that the process is not killed, my next problem is that the edges for the markers is not also visible in the cloud window which is showing the pointclouds.
[image]

pedghz, 2019-02-10:
Why the cloud window is showing sth like that?!

pedghz, 2019-02-11:
what does this line mean?
"Failed to find match for field 'ring'."

zeroAska, 2019-05-03:
Same issue here. Any suggestions?

hustlli, 2019-05-29:
> what does this line mean?
> "Failed to find match for field 'ring'."
> 
> P.S. I am using robosense LiDAR.
The robosense topic /rslidar_points does not indicate which ring an point belongs to (while velodyne does). For a 16 ring robosense lidar which publishes 2016 x 16 points each frame, every 2016 points belong to the same ring. You can manually set this attribute, Like this:

        int pt_index = 0;
        float ring_index = 0.0;
        pcl::PointCloud<myPointXYZRID>::Ptr new_cloud ( new pcl::PointCloud<myPointXYZRID>);
        for (pcl::PointCloud<myPointXYZRID>::iterator pt = point_cloud_raw.points.begin(); pt < point_cloud_raw.points.end(); pt++)
        {
                // 16 rings, from 0 to 15
                ring_index = pt_index/2016;
                pt->ring = floor(ring_index);
                pt_index = pt_index + 1;

                // filter NaN points
                if  (isnan(pt->x) == false)
                {
                        myPointXYZRID PointT;
                        PointT.x = pt->x;
                        PointT.y = pt->y;
                        PointT.z = pt->z;
                        PointT.intensity = pt->intensity;
                        PointT.ring = pt->ring;
                        PointT.range = pt->range;
                        new_cloud->points.push_back(PointT);
                }
        }

bulinx, 2019-10-19:
Hello, I have encountered the same problem. Can you solve it?

SiddhantNadkarni, 2020-04-28:
> what does this line mean?
> "Failed to find match for field 'ring'."

It basically means your LiDAR doesn't publish point cloud with a "ring" field. It shouldn't  be a major issue from my experience, I used a kinect to calibrate successfully which don't publish point cloud containing a "ring" field.