ankitdhall/lidar_camera_calibration #82: Aruco_mapping appears to recognize only one aruco marker. When two are recognized, the Mono8 interface turns gray.

hunkyu, 2019-02-13:
Aruco_mapping appears to recognize only one aruco marker. When two are recognized, the Mono8 interface turns gray.
[image]
[image]
[image]

hunkyu, 2019-02-13:
I feel that the setup is fine, but at the same time there are two arcco makers that will crash.

BmlmnnsDev, 2019-02-15:
Try to use the `--clock` argument when playing the rosbag. And use `<param name="/use_sim_time" value="true"/>`. I had the same issue. That configuration seemed to resolve that.

hunkyu, 2019-02-15:
@BmlmnnsDev  But without rosbag play, I am using the topic of the camera obtained online.

hunkyu, 2019-02-15:
@BmlmnnsDev  What do you mean by letting my rosbag record lidar and camera topic and then use --lock?

hunkyu, 2019-02-16:
thank you ! @BmlmnnsDev

closed 2019-02-20 by hunkyu