ankitdhall/lidar_camera_calibration #83: Why is the cloud interface point abnormal?

hunkyu, 2019-02-16:
[image]
my config file:
[config_file.txt](https://github.com/ankitdhall/lidar_camera_calibration/files/2872167/config_file.txt)
1792 930
-2.5 2.5 
-4.0 4.0 
0 2.5
0.005
2
0
923.49779 0.0 861.95580 0.0
0.0 921.20685 435.44906 0.0
0.0 0.0 1.0 0.0
100
1.57 -1.57 0

hunkyu, 2019-02-17:
[image]
[image]
I changed the config configuration again, but there are still a lot of invalid points.
my config file:
1792 930
-3 3
-1.0 4.0
0 2.5
0.005
2
0
923.49779 0.0 861.95580 0.0
0.0 921.20685 435.44906 0.0
0.0 0.0 1.0 0.0
100
1.57 1.57 0

closed 2019-02-22 by hunkyu

ykk136, 2020-12-22:
Hello! Would you mind showing me your "find_transform.launch" file? Thank you..