ankitdhall/lidar_camera_calibration #84: Final ypr is varying widely

BmlmnnsDev, 2019-02-21:
[image]

I use a recorded bag file and experienced an instable behavior of the calibration method. Using the identical bag file and identical parameters the method delivers varying results. 

config_file.txt
```
1920 1216
-100 100
-100 100
0 4.7
0.01
2
0
2107.007080 0 1034.812492 0 
0 2157.546875 515.748594 0 
0 0 1 0
100
1.57 -1.57 0.0
```

marker_coordinates.txt

```
2
100.0
70.0
5.0
5.0
30.0
100.0
70.0
5.0
5.0
30.0
```

zed_left_uurmi.ini

```
# Camera intrinsics

[image]

width
1920

height
1216

[camera]

camera matrix
2185.564228 0 1023.330735
0 2188.853599 515.806255 
0 0 1.0

distortion
-0.188516 0.106128 0.004597 0.005068 0.000000

rectification
1.000000 0.000000 0.000000
0.000000 1.000000 0.000000
0.000000 0.000000 1.000000

projection
2107.007080 0 1034.812492 0 
0 2157.546875 515.748594 0 
0 0 1 0
```

Result 1:

[image]

Result 2:
[image]

For the last case:

points.txt
```
8
-0.287939 -0.587037 4.17691
-0.0106564 -0.39377 4.13487
-0.39765 0.543588 4.18022
-0.948976 0.166455 4.23781
0.569169 -0.554285 4.1282
0.909962 -0.464728 4.14648
0.505134 0.451371 4.14054
0.18765 0.350935 4.17093
-0.643128 -1.04079 3.01878
-0.0559926 -0.661605 2.9802
-0.598641 0.17835 2.97742
-1.18578 -0.200838 3.016
0.223837 -0.985719 2.98987
0.889473 -0.770182 2.96814
0.580854 0.180763 2.94677
-0.0847825 -0.0347736 2.9685
```
transform.txt
```
3=(519.588,297.005) (404.96,478.401) (238.47,365.183) (351.442,186.267) Txyz=-0.909586 -0.260523 3.00583 Rxyz=-0.616471 1.89678 -1.93183 
9=(1285.19,324.969) (1217.91,534.697) (1007.38,466.432) (1075.2,257.711) Txyz=0.167144 -0.163377 2.96676 Rxyz=-0.357585 2.03932 -2.08965 
```

Can somebody help me out ?
Thanks in advance.

zhangyuran-gg, 2024-11-19:
> -2

hello,do you have any new conclusions about this issue?

TillBeemelmanns, 2024-11-19:
We now use Matlab Calibration Toolbox to calibrate LiDAR-Camera sensor setups.