ankitdhall/lidar_camera_calibration #94: Only Mono8 window visible

aroongta, 2019-04-23:
Hello,
I tried to use the package with a usb cam and velodyne 16 lidar. I can only see the mono8 window popup, cannot see the point cloud and the polygon window.
There are no errors displayed on the terminal.
[image]
Could someone please look into it and pint out potential reasons for the issue.
Thank you

Elkulas, 2019-04-24:
If you use the rosbag data to calibrate, be aware of uncomment the `<param name="/use_sim_time" value="true"/>` and add `--clock` while rosbag play.
If it is the online calibration, the process should continued

aroongta, 2019-04-24:
Thank you for your response @Elkulas.
It is online calibration (I'm not using a rosbag). Still only the mono8 window is popping up.

Elkulas, 2019-04-25:
@aroongta I got the same problem before I use the rosbag. Probably it is the launch file issue. Why not try rosbag record and calibrate.

Rrycbarm, 2019-06-07:
I'm having the same issue, have you managed to fix it? @aroongta

dkhanna511, 2020-11-06:
I'm having the same issue. were you able to fix it? @aroongta

yulan0215, 2020-12-08:
> I'm having the same issue. were you able to fix it? @aroongta

I have same issue, do you have any idea about it?

miriamrebekah, 2021-04-16:
@dkhanna511 @yulan0215 part of the issue seems to be in ```intensityByRangeDiff();``` I had to make a local version that fixed the first line of the function so that ```rings()``` is called on 32 instead of 16.

```std::vector <std::vector<myPointXYZRID *>> rings(32);```

After that I have had issues with the ```cv.waitKey(0);``` in the corners.cpp file which I am still working through.