ankitdhall/lidar_camera_calibration #102: Cannot detect markers

zgxsin, 2019-09-16:
Here is my setup: 
[image]

Here is my cloud output:
[image]

The point cloud is weird. 

In my terminal, i got this:
[image]

Anyone having the similar issue?

xingxiuhua, 2019-10-03:
Hello, can you help me to realize the marker window in the camera window?thank you very much!

dkhanna511, 2020-11-05:
HI, after running the find_transform.launch file, my camera and lidar window is not opening. I have not created the setup as of now but I want to anything pops up when there is no marker at first.

VictorEscribano, 2021-03-15:
It can be several things:
1. the Arucos are not in ascending order from left to right
2. The arucos are too small, try to print the arucos used in the paper, the number 26 annd 582 from the original dictionary
Hope it helped :)

Z-Jeff, 2022-12-01:
> 2\. 26

How can I download the 26 and 582 arucos?

louislelay, 2024-07-24:
> > 2. 26
> 
> How can I download the 26 and 582 arucos?

Hi @Z-Jeff, here’s the PDF of the markers. You can create your own using: https://chev.me/arucogen/

[ArUco_original_582.pdf](https://github.com/user-attachments/files/16357626/ArUco_original_582.pdf)
[ArUco_original_26.pdf](https://github.com/user-attachments/files/16357627/ArUco_original_26.pdf)