ankitdhall/lidar_camera_calibration #104: Fatal error: lidar_camera_calibration/marker_6dof.h: No such file or directory

NoRomantic, 2019-09-22:
When I run the commond 'catkin_make -DCATKIN_WHITELIST_PACKAGES="aruco_mapping;lidar_camera_calibration"
' as README said, the following error occured. What should I do, please?

`home/.../calibrate_ws/src/lidar_camera_calibration/src/find_velodyne_points.cpp:34:50: fatal error: lidar_camera_calibration/marker_6dof.h: No such file or directory
compilation terminated.
lidar_camera_calibration/CMakeFiles/find_transform.dir/build.make:62: recipe for target 'lidar_camera_calibration/CMakeFiles/find_transform.dir/src/find_velodyne_points.o' failed
make[2]: *** [lidar_camera_calibration/CMakeFiles/find_transform.dir/src/find_velodyne_points.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:518: recipe for target 'lidar_camera_calibration/CMakeFiles/find_transform.dir/all' failed
make[1]: *** [lidar_camera_calibration/CMakeFiles/find_transform.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed`

NoRomantic, 2019-09-24:
After three days of google and baidu, I finally make it build with no error. The key problem here is that the recommended build commads
 `catkin_make -  DCATKIN_WHITELIST_PACKAGES="aruco_mapping;lidar_camera_calibration`
went wrong, because before building `aruco_mapping` you must build `lidar_camera_calibration` first(but still, `aruco_ros` should be built before `lidar_camera_calibration`). So the right build order is `aruco_ros`, `lidar_camera_calibration`, and last the `aruco_mapping`.

And I met  a problem when I build lidar_camera_calibration package,  i.e. `Does the package have a <build_depend> on message_generation in its package.xml?`. I found it was caused by the ros message mechanism, however, I don't quite understand this. So if you are familiar with Chinese, I recommon you have a look at this link [https://www.cnblogs.com/lizhensheng/p/11117473.html](url).

Wish this post may help you. Thanks.

ankitdhall, 2019-09-24:
@karnikram suggested adding that to the README.md. Maybe he can be of help with that one? @karnikram

closed 2019-09-26 by NoRomantic