ankitdhall/lidar_camera_calibration #109: Unable to calibrate

androclassic, 2019-11-30:
Can anyone help me, I have been trying to calibrate a zed camera with a velodyne lidar using this but I cannot reproduce what I see in the usage video, basically in my case the cloud window is black and the mono window never closes. I have recorded a video to better show my problem.
https://photos.app.goo.gl/qiJocrEXPHCgC2oR8

SiddhantNadkarni, 2020-04-26:
A good way to verify calibration can be by projecting the LiDAR points on the Camera image plane to and then visually verify. Or Intergrate RGB information from Image into the Point Cloud and then visually verify.

Gold3nFox, 2022-04-08:
> A good way to verify calibration can be by projecting the LiDAR points on the Camera image plane to and then visually verify. Or Intergrate RGB information from Image into the Point Cloud and then visually verify.

How can i project lidar points on the camera ? can you help me on #143