ankitdhall/lidar_camera_calibration #112: monocular camera image with Lidar Points

SinghiDivyanshu, 2020-01-13:
hi,
what is the next step after finding the rigid body transformation using Velodyne LIDAR and monocular camera. my RMSE value is 0.1636.

so how to find the image with LIDAR point cloud on it.

SiddhantNadkarni, 2020-04-26:
Hi @SinghiDivyanshu, once you have the rigid body 4x4 transformation matrix you'll have to perform the following steps:
1. Transform the point cloud captured from LiDAR to Camera Co-ordinate system by using the 4x4 Rigid Body Transformation matrix obtained from this package.
2. Project the Points from Camera Co-ordinate system to image plane by using 3x4 Camera Instrinsic / Projection Matrix which contains camera info such as focal lengths etc. 

Let me know if you have doubts.

AnthonyLadewig, 2020-11-02:
Hi @SiddhantNadkarni ,

How you do transform the point cloud to the camera coordinate system and then project it?
I don't know what packages or code to use.

dkhanna511, 2020-11-05:
Can you please mention what changes you had to make in the launch and lidar_camera_calibration.yaml to get it running, my topic names are different and I'm not able to understand which change has to be made where.

AnthonyLadewig, 2020-11-05:
@dkhanna511 
In the launch file, you have to uncomment the code under aruco mapping and then change the remap topic.
<remap from="image_raw" to="zed/left/image"/>
Change the zed/left/image to your camera's topic.

In the yaml file, do the same.

To find your camera's topics, run the launch for your camera and do 'rostopic list' in the terminal.

dkhanna511, 2020-11-05:
@AnthonyLadewig  
Hi, thanks for the response. I did these changes, Now I'm facing an openCV issue. that is mentioned in #131. Do you have idea about how to solve that? Have you encountered this issue?

AnthonyLadewig, 2020-11-05:
@dkhanna511 
Unfortunately I have not had this problem and do not know how to fix it. Sorry

dkhanna511, 2020-11-06:
Hi, I ran the whole thing on another laptop and it works. Apparently there are some openCV conflict issues. Now, after running the find_transform.launch file, only mono8 window is opening. I have not created the setup as of now but I want to know if the polygon window and point cloud window pops up when there is no marker at first.

dkhanna511, 2020-11-06:
> @dkhanna511
> Unfortunately I have not had this problem and do not know how to fix it. Sorry

@AnthonyLadewig 
 I ran the same process on another laptpo and it works. Apparently there are some OpenCV conflict issues. After running the find_transform.launch file, only mono8 window is opening. I have not created the setup as of now but I want to know if the polygon window and point cloud window pops up when there is no marker at first.

AnthonyLadewig, 2020-11-06:
@dkhanna511 from my experience, the polygon/point cloud window doesn't come up if the lidar isn't running.

dkhanna511, 2020-11-09:
> @dkhanna511 from my experience, the polygon/point cloud window doesn't come up if the lidar isn't running.

@AnthonyLadewig The lidar is running. I can see the point cloud fees on RVIZ

AnthonyLadewig, 2020-11-10:
@dkhanna511 Im not sure what the problem might be then.
Is the calibration software actually subscribing to the lidar topic?
What lidar are you using btw?

dkhanna511, 2020-11-11:
@AnthonyLadewig  I was using VLP16. I had to install everything from scratch and now all the 3 windows come up.