ankitdhall/lidar_camera_calibration #113: The polygon and cloud windows have no data

hugomarques5, 2020-02-28:
Hi

I am currently trying to calibrate a Leishen 3D LIDAR 
(http://en.leishen-lidar.com/product/leida/MX/15d44ea1-94f5-4b89-86eb-f5a781b04078.html) and a Intel stereo camera (https://store.intelrealsense.com/buy-intel-realsense-depth-camera-d415.html). Both are implemented in a bobcat and in fixed positions, however the camera is pointing further down and the LIDAR is pointing forward, which means that the data of both have little intersection.

When testing the package, I got to the point where you provide me with an initial [R | t] between the camera and the various ArUco markers, but when you open the "polygon" and "cloud" windows no data appears. I saw the youtube video tutorial and I expected to see something similar but it didn't.

Arucos are visible to the camera but are not fully detected by the LIDAR. Could the problem lie in this camera/LIDAR low data intersection? Or is it enough that the Arucos are visible to the camera and not to LIDAR? Can you tell me if there is another reason for this problem?

I appreciate any help!

[image]

kunalchelani, 2020-03-02:
Hi @hugomarques5 . As far as I remember, the 3D coordinates of the corner points of the boards on which the aruco markers are stuck are needed to be in the lidar feed. This is so that we have 3D-3D correspondences of the corner points in both - camera's and lidar's frame of reference, which further allows for calculation of the [R|t].

hugomarques5, 2020-03-02:
Thank you @kunalchelani ! 

In order to solve the intersection problem, another camera was attached so that both cards and Arucos were fully visible to LIDAR and the camera. However, after changing all the configuration parameters necessary for the use of the new camera, the error of not having data in the windows remained (see the images). Can you tell me what is the reason for this problem?

[image]

[image]

hugomarques5, 2020-03-02:
Thank you @kunalchelani !

jiqirenno1, 2021-07-07:
I have the same problem，Has the problem been solved ？ @hugomarques5