ankitdhall/lidar_camera_calibration #115: TF to MSG: Quaternion Not Properly Normalized

roylu307, 2020-03-05:
I ran into this problem either I was calibrating online or with rosbag. with rosbag I used --clock and uncommented the sim time parameter in the launch file. It seems like the problem with TRANSFORM. has anyone run into this problem and solved it?

The process was killed but still detecting the markers as indicated by the terminal. Only the mono8 window was shown.

SUMMARY
========

PARAMETERS
 * /aruco_mapping/calibration_file: /home/roy/spinnak...
 * /aruco_mapping/marker_size: 0.2
 * /aruco_mapping/num_of_markers: 2
 * /aruco_mapping/roi_allowed: False
 * /aruco_mapping/space_type: plane
 * /lidar_camera_calibration/camera_frame_topic: /camera_array/cam...
 * /lidar_camera_calibration/camera_info_topic: /camera_array/cam...
 * /lidar_camera_calibration/velodyne_topic: /velodyne_points
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /use_sim_time: True

NODES
  /
    aruco_mapping (aruco_mapping/aruco_mapping)
    find_transform (lidar_camera_calibration/find_transform)

ROS_MASTER_URI=http://localhost:11311

process[aruco_mapping-1]: started with pid [9387]
process[find_transform-2]: started with pid [9388]
[ INFO] [1583427817.673956045]: Calibration file path: /home/roy/spinnaker_ws/src/aruco_mapping/data/zed_left_uurmi.ini
[ INFO] [1583427817.674008534]: Number of markers: 2
[ INFO] [1583427817.674031962]: Marker Size: 0.2
[ INFO] [1583427817.674046015]: Type of space: plane
[ INFO] [1583427817.674061787]: ROI allowed: 0
[ INFO] [1583427817.674078689]: ROI x-coor: 1702043748
[ INFO] [1583427817.674093541]: ROI y-coor: 1702043748
[ INFO] [1583427817.674108455]: ROI width: 543649385
[ INFO] [1583427817.674121908]: ROI height: 1830839924
Size of the frame: 2048 x 1536
Limits: -2.5 to 2.5
Limits: -4 to 4
Limits: 0 to 2.5
Number of markers: 2
Intensity threshold (between 0.0 and 1.0): 0.05
useCameraInfo: 0
Projection matrix: 
[1390.7552, 0, 1039.3367, 0;
 0, 1390.4637, 768.77509, 0;
 0, 0, 1, 0]
MAX_ITERS: 100
Lidar_type: Velodyne
initial rotation: 1.57 -1.57 0
initial translation: 0 0 0
[ INFO] [1583427817.678097510]: Calibration data loaded successfully
[ INFO] [1583427817.682240179]: Reading CameraInfo from configuration file
26=(800.428,504.513) (908.151,584.925) (827.571,693.099) (720.963,612.939) Txyz=-0.33583 -0.256717 2.06047 Rxyz=-0.797583 -1.64809 1.66158 
582=(1258.5,584.203) (1366.49,662.215) (1291.55,772.116) (1182.37,694.123) Txyz=0.348411 -0.138969 2.05771 Rxyz=-0.942141 -1.68777 1.64306 
[ INFO] [1583427819.849306515, 1582565468.187454146]: First marker with ID: 26 detected
**_[ WARN] [1583427819.976807360, 1582565468.328572356]: TF to MSG: Quaternion Not Properly Normalized
Error:   TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "camera_position" from authority "unknown_publisher" because of an invalid quaternion in the transform (-0.000000 0.000000 0.000000 0.000000)
         at line 257 in /tmp/binarydeb/ros-kinetic-tf2-0.5.20/src/buffer_core.cpp_**
[ INFO] [1583427819.992628794, 1582565468.338658159]: Velodyne scan received at 1.58257e+09
[ INFO] [1583427819.992728401, 1582565468.338658159]: marker_6dof received at 1.58257e+09

Initial Rot 0.000796274    -0.999999 -0.000796274
           0  0.000796274           -1
           1  0.000796274  6.34053e-07
26=(800.307,504.395) (908.168,584.836) (827.367,693.065) (720.831,612.849) Txyz=-0.335721 -0.256637 2.05876 Rxyz=-0.797163 -1.64592 1.66389 
582=(1258.47,584.208) (1365.97,662.36) (1291.43,772.133) (1182.55,694.093) Txyz=0.348337 -0.13893 2.05793 Rxyz=-0.955682 -1.69736 1.63967 
_**[find_transform-2] process has died [pid 9388, exit code -11, cmd /home/roy/spinnaker_ws/devel/lib/lidar_camera_calibration/find_transform __name:=find_transform __log:=/home/roy/.ros/log/ad19d36a-5f00-11ea-a683-0c54156ec1f2/find_transform-2.log].
log file: /home/roy/.ros/log/ad19d36a-5f00-11ea-a683-0c54156ec1f2/find_transform-2*.log**_
26=(800.349,504.462) (908.172,584.763) (827.388,693.006) (720.68,612.812) Txyz=-0.335738 -0.256709 2.05881 Rxyz=-0.80504 -1.6445 1.66828 
582=(1258.46,584.267) (1365.87,662.442) (1291.38,772.121) (1182.63,694.08) Txyz=0.348459 -0.13895 2.05881 Rxyz=-0.957674 -1.69839 1.63997 
26=(800.341,504.406) (908.142,584.773) (827.349,693.016) (720.641,612.78) Txyz=-0.335732 -0.25669 2.05849 Rxyz=-0.803772 -1.64507 1.66758 
582=(1258.43,584.296) (1365.52,662.404) (1291.4,772.14) (1182.72,694.048) Txyz=0.348398 -0.138958 2.05879 Rxyz=-0.966515 -1.70329 1.63824

closed 2020-03-05 by roylu307

dejongyeong, 2020-06-11:
@roylu307 I got the same issue too. Do you have any idea how to solve it? Thanks

crankler, 2020-06-18:
Have you all solve this problem?

miriamrebekah, 2021-05-03:
@roylu307 , did you every get a solution to this issue?

Wuxinxiaoshifu, 2022-03-22:
i also meet this problem! Has somebody solved this?

0NNS0, 2022-12-06:
我也遇到了这个问题，然后得到了解决。
我遇到这个问题的情况是：我在电脑上对录制的rosbag进行标定。
解决问题的情况是：我在电脑上对传感器进行实时的标定。
原因我认为是：如果对录制的rosbag进行标定，雷达的时间戳是录制时的时间戳，而aruco mapping的时间戳是当前运行程序时的时间戳，所以TF不会进行转化。
所以为了避免此问题的出现，必须对传感器进行实时的标定，而非录制bag后再进行标定。

Lin239522, 2023-02-25:
如果录制bag包后，标定内参，再将内参设置为ros节点发布，是可以的嘛

> 我也遇到了这个问题，然后得到了解决。 我遇到这个问题的情况是：我在电脑上对录制的rosbag进行标定。 解决问题的情况是：我在电脑上对传感器进行实时的标定。 原因我认为是：如果对录制的rosbag进行标定，雷达的时间戳是录制时的时间戳，而aruco mapping的时间戳是当前运行程序时的时间戳，所以TF不会进行转化。 所以为了避免此问题的出现，必须对传感器进行实时的标定，而非录制bag后再进行标定。