ankitdhall/lidar_camera_calibration #127: I am not able to build the repository on my ROS-melodic. please help.

chaitradeepika, 2020-08-09:
catkin_make -DCATKIN_WHITELIST_PACKAGES="aruco_mapping;lidar_camera_calibration"
Base path: /home/chaitra/catkin_ws
Source space: /home/chaitra/catkin_ws/src
Build space: /home/chaitra/catkin_ws/build
Devel space: /home/chaitra/catkin_ws/devel
Install space: /home/chaitra/catkin_ws/install
####
#### Running command: "cmake /home/chaitra/catkin_ws/src -DCATKIN_WHITELIST_PACKAGES=aruco_mapping;lidar_camera_calibration -DCATKIN_DEVEL_PREFIX=/home/chaitra/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/chaitra/catkin_ws/install -G Unix Makefiles" in "/home/chaitra/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/chaitra/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/melodic
-- This workspace overlays: /opt/ros/melodic
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.9", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/chaitra/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python3 (found version "3.6.9") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.26
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- Using CATKIN_WHITELIST_PACKAGES: aruco_mapping;lidar_camera_calibration
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - lidar_camera_calibration
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'lidar_camera_calibration'
-- ==> add_subdirectory(lidar_camera_calibration)
CMake Deprecation Warning at /opt/ros/melodic/share/ros/core/rosbuild/rosbuild.cmake:7 (cmake_policy):
  The OLD behavior for policy CMP0011 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  lidar_camera_calibration/CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/melodic/share/ros/core/rosbuild/rosbuild.cmake:16 (cmake_policy):
  The OLD behavior for policy CMP0002 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  lidar_camera_calibration/CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/melodic/share/ros/core/rosbuild/rosbuild.cmake:18 (cmake_policy):
  The OLD behavior for policy CMP0003 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  lidar_camera_calibration/CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/melodic/share/ros/core/rosbuild/rosbuild.cmake:20 (cmake_policy):
  The OLD behavior for policy CMP0005 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  lidar_camera_calibration/CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/melodic/share/ros/core/rosbuild/rosbuild.cmake:23 (cmake_policy):
  The OLD behavior for policy CMP0011 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  lidar_camera_calibration/CMakeLists.txt:2 (include)

CMake Error at /home/chaitra/catkin_ws/devel/share/velodyne_pointcloud/cmake/velodyne_pointcloudConfig.cmake:173 (message):
  Project 'lidar_camera_calibration' tried to find library
  'velodyne_rawdata'.  The library is neither a target nor built/installed
  properly.  Did you compile project 'velodyne_pointcloud'? Did you
  find_package() it before the subdirectory containing its code is included?
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  lidar_camera_calibration/CMakeLists.txt:12 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/chaitra/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/chaitra/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

closed 2020-08-11 by chaitradeepika

dkhanna511, 2020-11-02:
I'm facing a similar issue. Could you please give the solution here as well?

vefak, 2021-09-08:
aq ya madem buldun cevabı paylaş